openhardwaremonitor/openhardwaremonitor #1105: Support for MSI Z370 Gaming Plus (MS-7B61)

ElspethMT, 2018-05-22:
Is there a plan to add support for Z370 motherboards? I need something that will allow me to track the fans and temps and I really like this program. Sadly, it does not seem to recognize my motherboard sensors. 

[image]

DarthDestroyer, 2018-05-22:
could you add a report, please?

i bet, there is already a pull request for that chipset

ElspethMT, 2018-05-22:
Attached is the report.
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/2028956/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2018-05-23:
1. outdated Version: http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

2. Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0xD352 at 0x4E/0x4

3. Take a Look at #884

ElspethMT, 2018-05-23:
I'm sorry, I'm new to this. Does that mean that I have to edit the code to make it work?

DarthDestroyer, 2018-05-23:
right now, yes!
It is a temporal solution and you can help test that Pull Request.

ElspethMT, 2018-05-23:
OK, I have never done any programing so I have more dumb questions.

1. Which file do I need to edit?
2. What program do I need to install in order to edit the code?

DarthDestroyer, 2018-06-28:
The easiest way would be:

get visual studio 
get the Sourcecode
edit the files which are listed in #884 (green=add / red=remove)
compile/run it and you are done ;-)

Bonilla, 2018-09-09:
Hello, I also have an MSI Z370 board and I have followed the advice of @DarthDestroyer and compiled OpenHardwareMonitor myself. It seems to be working!

However, I am having a hard time telling which of the temperatures (Nuvoton NCT6795D > Temperatures) represents which sensor on the mainboard. I am also wondering why there are six fan controls even though the mainboard is only equipped with 5 fan sockets (1x CPU, 3x chassis, 1x pump). Any help you would be greatly appreciated! Thanks.

[image]

martynsmall, 2018-09-09:
Chips such as the Nuvoton NCT6795D you have are present on many boards (my MSI Z270 has the same one), but not all boards will make use of all the available sensors/outputs.  That chips supports six fans, but as your board only has five then the last is simply unused.  However, as OpenHardwareMonitor doesn't know the specifics about your board it has no choice other than to just display everything offered by the chip.  You can either hide the additional one through the program itself (right-click it), or add support for your actual board in code and only map five of the fans.

The same is true for the temperature sensors, the chip supports a lot more than are actually used.  On mine `Temperature #2` is the System (I checked this against the MSI utility), and `Temperature #1` seems to be something near the CPU - it increases with the CPU but doesn't get as high and takes longer to change.  The remaining ones on mine are unused and have values that are either constant or only change by 1 degree.  Given how low `#3`, `#4` and `#6` are for you (`#5` will likely be hidden as it's 0), I'd say these aren't connected to anything either.

I've added support for a couple of boards on my fork if you're interested in seeing how to do the same and given these two are both MSI, albeit different chips, and have the same sensors I'd be inclined to say yours are the same too but I can't guarantee it!

Z270 PC Mate: 95ebe0ac630e04778ff768fc9b1830ff8f482229
B360M PRO-VDH: 30099a54cabf036bdb8bdb964921644d95818c4a

Bonilla, 2018-09-09:
Great, thank you for explaining everything! I went into the BIOS and it seems sensor #1 represents "MOS" and sensor #2 "System".