openhardwaremonitor/openhardwaremonitor #1108: Incorrect total VRAM for Nvidia 1070ti

mtreis86, 2018-06-09:
My hardware report is coming back with the wrong total VRam on my 1070. Shows 4gigs where the 1070 has 8.

DarthDestroyer, 2018-06-09:
could you upload a report?

mtreis86, 2018-06-09:
I submitted a report to you through OHM Submit. If you would like another copy please send me your e-mail.

mtreis86, 2018-06-09:
I may just be reading the report wrong. It clearly displays GPU Memory Total correctly. What is the GPU Memory (/nvidiagpu/0/clock/1) line in regards to?

DarthDestroyer, 2018-06-09:
attach it here, please!

clock means clock ;-)

mtreis86, 2018-06-09:
[MTReis86.OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/2087094/MTReis86.OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2018-06-09:
looks fine for me?!

you could use the latest Version, but not really needed for you!

mtreis86, 2018-06-09:
Okay I must be misunderstanding the file then. I will close this issue. Thanks.

closed 2018-06-09 by mtreis86