openhardwaremonitor/openhardwaremonitor #1109: Different Profiles?

Fround1, 2018-06-15:
Can you make a profile? like one for gaming and one for just browser, fans at 80%+ while just watching a youtube video gets pretty annoying. and i can just change the speeds but i have 6 fans (7 with gpu) so its a long and boring process that takes 3-4 min. Heres a photo for better understanding.
[image]

DarthDestroyer, 2018-06-15:
#762 could that be, what you want?

Fround1, 2018-06-15:
yes it is. how can i add that?

DarthDestroyer, 2018-06-15:
The easiest way would be:
1. get visual studio
2. get the Sourcecode 
3. edit the files which are listed here (green=add / red=remove)
4. compile/run it and you are done ;-)

If that PR will get more tested, it will maybe accepted and possible in a new release

Fround1, 2018-06-17:
is there any way to get the full pack with hardware monitor and the sourcecode in it so it works right after download since i have no idea what im doing. ive been searching for 2 days with no results. i have minimum experience with visual studio.