openhardwaremonitor/openhardwaremonitor #1114: All core temperatures missing for i7-7820X

syl20bou, 2018-06-27:
When I use openHardwareMonitor, despite recent updates to support Skylake, I get zeros for all 8 temperature sensors and package temperature as well. Other counters have proper values such as load values.

I use version 0.8.0.5 of the code. That version reads temperatures properly on my i7-4850HQ.

highpec, 2018-07-15:
Same on i9-7940x, MSI x299m Carbon

Nuvoton NCT6795

mrstandu33, 2018-07-15:
same here, you can download source files and compile it, it will normally work (I've seen a commit fixing this bug, but the latest release was before fix, so you have to compile it yourself (didn't try, as I don't have what it need to compile)

mrstandu33, 2018-07-15:
I just compiled it, everything is working fine now (check screens). You can DL files here : https://we.tl/SEClsxVcIa

Hope you'll enjoy :p
[image]

highpec, 2018-07-16:
Very nice.  It works now...!

mrstandu33, 2018-07-16:
mark as solved now ? 🤔

closed 2018-07-16 by syl20bou