openhardwaremonitor/openhardwaremonitor #1116: External USB HDD and USB Stick not detected when inserted.

JamesLDoty, 2018-07-08:
I experimented by plugging in and removing old (6-8 years) and new (<6 months) USB hard drives and USB flash drives.  In all cases, the HardwareAdded and HardwareRemoved events were not fired, and they did not not show up in the scan.  They did show up if I shut the application down and restarted it.

Windows 10 (1803)
ASUS P6T Mobo
Intel Core i7 920

Screen shot without after adding the external HDD:
[image]

and with after restarting:
[image]

JamesLDoty, 2018-07-08:
PS: I'm using OpenHardwareMonitor 0.8.0 Beta

JamesLDoty, 2018-07-08:
> With OHM you can try the Reset Button if it detects them after plugging in?

But I have to do that manually, right?  If I want to create an application, it should get some notification that the hardware configuration has changed dynamically.

 

Jim

-----------------------------

JamesLDoty, 2018-07-08:
BTW: it’s a great app.

 

Jim

-----------------------------

DarthDestroyer, 2018-07-08:
the first question is, if it is then recognized ;-)

after that, we know more

JamesLDoty, 2018-07-09:
I’ll try some things this morning.

 

Jim

-----------------------------

JamesLDoty, 2018-07-09:
I didn’t say it in my previous post, but calling the Open() and Close() methods on each update did work.  It just slowed my system down to a crawl.

 

Jim

-----------------------------

DarthDestroyer, 2018-07-10:
you could fetch the "usb connected" event in your code and add there a close and reopen!

JamesLDoty, 2018-07-10:
Dan:

 

Tried that using WMI.  Did the following:

 

ManagementEventWatcher watcher = new ManagementEventWatcher();

WqlEventQuery query = new WqlEventQuery("SELECT * FROM Win32_VolumeChangeEvent WHERE EventType = 2 or EventType = 3");

watcher.EventArrived += Watcher_EventArrived;

watcher.Query = query;

watcher.Start();

 

It worked, I got a notification in Watcher_EventArrived, and called Close then Open.  But I couldn’t eject the device.  When I tried to, I kept getting the following:

 

 

 

If I turned off monitoring HDD, then I could immediately eject the device.  I assume OHM opens a handle to each device and keeps it open, which would prevent ejection.

 

Jim

-----------------------------

DarthDestroyer, 2018-07-10:
oh... yes! i forgot that!
There is a Issue already opened for that... 

Maybe it could be possible to catch the eject event and close and pause for that?

JamesLDoty, 2018-07-10:
That doesn’t work because I can’t eject it while OHM is running, so there is no eject event until I shut down OHM.

 

It sounds like you’re working on that.  I just published an article in CodeProject about OHM and how to use it.  I discussed this issue.  If you are working on it, and you think you have a fix, let me know and I’ll test it out for you, and update the article.

 

Jim

-----------------------------