openhardwaremonitor/openhardwaremonitor #1120: No fan speed displayed for i5-4200U

EtienneBasilik, 2018-08-01:
I have an Acer Aspire V7 582P with an Core i5-4200U, and the fan speed are not showing up. I have an overheat problem and would like to see if my fans are working nominally. I've attached the report. Thanks!
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/2249269/OpenHardwareMonitor.Report.txt)

aldfrarug, 2018-08-01:
Your bios is not supported by openhardwaremonitor (looks like at least). Maybe hwmonitor by cpuid works.

DarthDestroyer, 2018-08-01:
could you try the latest version? > http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

The last time i saw 
`Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0x8587 at 0x2E/0x2F

Chip ID: Unknown ITE with ID 0x8587 at 0x2E/0x2F

Chip ID: Unknown SMSC with ID 0x8587 at 0x2E/0x2F`
is atleast a year ago!