openhardwaremonitor/openhardwaremonitor #1126: New B450  mobo (Asrock B450M Pro 4)

HarisHashim, 2018-08-11:
Attached the saved report and screenshot showing actual value for vcore voltage and CPU temp

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/2280612/OpenHardwareMonitor.Report.txt)

[image]

DarthDestroyer, 2018-10-26:
For Ryzen Support, there are several Pull Requests

For your VCore, it seems to be exactly split by half

HarisHashim, 2019-01-02:
I just noticed today that sometime Open Hardware Monitor report motherboard temperature less than Asrock monitoring tool. Like 50C and Asrock monitoring tool is 86C and the fan is blazing.