openhardwaremonitor/openhardwaremonitor #1128: PRIME B350-PLUS AMD Ryzen 7 2700X Pls Help problem with Software

Xiantrius, 2018-08-18:
hi i use openhardwaremonitor v8.0.0 beta and next v8.0.5 alpha on 2 PC's.
I am sure there are many other problems:
Unfortunately, I can not say if the other Ryzen CPUs have the same problem.
I give information now price and hope that it will give a fix.
Thanks in advance :).
####################################################
My PC:
ASUS PRIME B350-PLUS Mainboard
AMD Ryzen 7 2700X 8 Kerne and 16 Threads
16GB DDR4 Ram
SSD Samsung 970 PRO M.2 512GB
AMD Radeon R9 390 Series 8GB
Windows 10
####################################################
I Found Infos:
Name:
ASUS Prime B350-PLUS
AMD Ryzen 7 2700X
Load 
CPU Total
CPU Core 1 to 16

OpenHardwareMonitor v8.0.0 and v8.0.5 same problems
[image]

I Not Found Infos = Problem:
That Programm OpenHardwareMonitor can't find my CPU Temperature etc...
i missed infos on CPU Display ...
CPU Voltages
CPU Temperatures
CPU Fans
CPU Clocks
####################################################
My Reserve PC:
ASUS M5 A97 R2.0 Mainboard
AMD FX 8350 8 Kerne and 8 Threads
32GB DDR3 Ram
HDD 512GB
AMD Radeon RX 480 Series 8GB
Windows 7
####################################################
Here Work:
Last Test Openhardwaremonitor.exe.
I see all Info's
####################################################
I Hope please help fix problem in new OpenHardwareMonitor Version?

Best regards,
Xiantrius

DarthDestroyer, 2018-08-18:
Hey, it would be better if you add Reports instead of that Informations!
Attach them to this Issue ;-)

Ryzen is a little bit more to do, you would find some help in the pull requests!

Grüße

Xiantrius, 2018-08-18:
ok here my report

Open Hardware Monitor Report

--------------------------------------------------------------------------------

Version: 0.8.0.5

--------------------------------------------------------------------------------

Common Language Runtime: 4.0.30319.42000
Operating System: Microsoft Windows NT 10.0.17134.0
Process Type: 64-Bit

--------------------------------------------------------------------------------

Sensors

|
+- ASUS PRIME B350-PLUS (/mainboard)
|
+- AMD Ryzen 7 2700X (/amdcpu/0)
|  +- CPU Total      : 0.585938 0.289965  3.60049 (/amdcpu/0/load/0)
|  +- CPU Core #1    :   4.6875        0       22 (/amdcpu/0/load/1)
|  +- CPU Core #2    :        0        0   21.875 (/amdcpu/0/load/2)
|  +- CPU Core #3    :        0        0    3.125 (/amdcpu/0/load/3)
|  +- CPU Core #4    :        0        0     6.25 (/amdcpu/0/load/4)
|  +- CPU Core #5    :        0        0        2 (/amdcpu/0/load/5)
|  +- CPU Core #6    :        0        0  4.61538 (/amdcpu/0/load/6)
|  +- CPU Core #7    :        0        0   1.5625 (/amdcpu/0/load/7)
|  +- CPU Core #8    :        0        0    3.125 (/amdcpu/0/load/8)
|  +- CPU Core #9    :        0        0  14.0625 (/amdcpu/0/load/9)
|  +- CPU Core #10   :        0        0   1.6129 (/amdcpu/0/load/10)
|  +- CPU Core #11   :   4.6875  1.96078  12.6984 (/amdcpu/0/load/11)
|  +- CPU Core #12   :        0        0    3.125 (/amdcpu/0/load/12)
|  +- CPU Core #13   :        0        0   1.5625 (/amdcpu/0/load/13)
|  +- CPU Core #14   :        0        0        0 (/amdcpu/0/load/14)
|  +- CPU Core #15   :        0        0  1.96078 (/amdcpu/0/load/15)
|  +- CPU Core #16   :        0        0        6 (/amdcpu/0/load/16)
|
+- AMD Radeon (TM) R9 390 Series (/atigpu/0)
|  +- GPU Core       :      0.9      0.9    1.075 (/atigpu/0/voltage/0)
|  +- GPU Core       :      300      300   326.93 (/atigpu/0/clock/0)
|  +- GPU Memory     :      150      150     1500 (/atigpu/0/clock/1)
|  +- GPU Core       :       41       41       42 (/atigpu/0/temperature/0)
|  +- GPU Core       :        0        0        0 (/atigpu/0/load/0)
|  +- GPU Fan        :     1419     1411     1421 (/atigpu/0/fan/0)
|  +- GPU Fan        :       20       20       20 (/atigpu/0/control/0)

--------------------------------------------------------------------------------

Parameters

|
+- ASUS PRIME B350-PLUS (/mainboard)
|
+- AMD Ryzen 7 2700X (/amdcpu/0)
|
+- AMD Radeon (TM) R9 390 Series (/atigpu/0)

--------------------------------------------------------------------------------

Mainboard

SMBIOS Version: 3.1

BIOS Vendor: American Megatrends Inc.
BIOS Version: 4011

System Manufacturer: System manufacturer
[... 1049 lines cut ...]

AdapterIndex: 5
isActive: 0
AdapterName: AMD Radeon (TM) R9 390 Series
UDID: PCI_VEN_1002&DEV_67B0&SUBSYS_93951682&REV_80_4&1DA95F35&0&0019&06A
Present: 1
VendorID: 0x1002
BusNumber: 7
DeviceNumber: 0
FunctionNumber: 0
AdapterID: 0x87F8E180

AdapterIndex: 6
isActive: 0
AdapterName: AMD Radeon (TM) R9 390 Series
UDID: PCI_VEN_1002&DEV_67B0&SUBSYS_93951682&REV_80_4&1DA95F35&0&0019&07A
Present: 1
VendorID: 0x1002
BusNumber: 7
DeviceNumber: 0
FunctionNumber: 0
AdapterID: 0x87F8E180

Best regards,
Xiantrius

Xiantrius, 2018-08-18:
@DarthDestroyer ich kann ehrlich gesagt besser Deutsch.
Kannst du Deutsch?

DarthDestroyer, 2018-08-18:
@Xiantrius Das habe ich schon gesehen! Deswegen die "Grüße" ;-)

Wie gesagt, für den Ryzen gibt es schon einige Pull Requests um dahingehend zu helfen! 

Dein Mainboard selbst habe ich mir noch nicht angesehen! Auf der Arbeit und mit dem Handy ist das unpraktisch :-|

Xiantrius, 2018-08-18:
Mach dir keine umstände schau erst nach wenn du Zeit findest und nicht auf der Arbeit bist.
Es ist richtig es gibt für den Ryzen 7 1600 schon infos. Aber ich habe ja ein Ryzen 7 2700X also zurzeit den neusten AMD Prozessor den es zum verkauf gibt in moment.

Ich gehe aber mal davon aus das die gesamte neusten AMD und Intel CPU's serien und die neusten AM4/Intel Boards usw...stück für stück noch hinzugefügt wird.
Es wäre schön, wenn die nexte version eines Tages erscheint, dass ich sie mal antesten darf.
Vielen herzlichen dank in vorraus.

mit freundlichen grüssen,
Xiantrius

DarthDestroyer, 2018-08-18:
Ich würde es wenigstens probieren die #970 zu testen!

Anschließend könnte man von dort aus weiter suchen ;-)

Xiantrius, 2018-08-18:
Ich verstehe nicht ganz mit C# kenne ich mich überhaupt nicht aus.
Anders komme ich wenigstens ein bisschen mit Visual Basic klar.
Ich könnte höchstens ein fertiges projekt das mit C# geschrieben wurde kompilieren damit ich das programm mal testen kann. Kann man den Code von den Thread #970 nicht irgendwo downloaden?

DarthDestroyer, 2018-08-18:
In dem PR habe ich auch noch einen weiteren Link angegeben, wo ich das ein wenig näher beschrieben habe! Ist aber auch nicht weiter schwer, eigentlich ;)

Xiantrius, 2018-08-18:
So ich habe den link gefunden, das projekt heruntergeladen und versucht zu kompilieren
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/archive/master.zip

Er hat die DLL erstellt die exe komischerweise nicht, weil er irgend eine klassenbibliothek nicht finden kann. kein plan was er will, wenn man doch davon ausgehen muss, dass man ja das ganze Projekt herunterladen kann.

Naja wie dem auch sei ich hab gesehen das er wenigstens die OpenHardwareMonitorLib.dll Datei v0.8.0.2 erstellt hat.
Naja ich hab sie dann versucht auf mein VB 2017 Projekt versucht anzuwenden
Die OpenHardwareMonitorLib.dll habe ich als seid der v0.8.0.0 Beta als Verweis hinzugefügt wie immer wenn es sich um eine neue Version handelt. Zurzeit ist die v0.8.0.5 enthalten. Da diese bisher die aktuellste version zu sein scheint. Bei mein Reserve PC hab ich auch in ersten Post geschrieben ging das ohne probleme. Die Temperatur wurde ohne wenn und aber ausgelesen. Naja ist ja auch ein AMD FX 8350 der schon seid 23.10.2012 auf den markt kam und existiert schon einiger Zeit.
Naja ich hab das nur angeschnitten um zu zeigen, dass der Fehler nicht an mein Code liegt.

```
Imports OpenHardwareMonitor.Hardware

Public Class Form1

    Dim Computer As New Computer
    'Und noch mehr dazu
    'Weiteres Beispiel
        Private Function GetCoreTemp(ByVal Core As Integer) As String
        For Each Hardware In Computer.Hardware 'Jede einzelne Hardware heraussuchen
            Hardware.Update() 'Die Werte aktualisieren
            If Hardware.HardwareType = OpenHardwareMonitor.Hardware.HardwareType.CPU Then 'Da wir die CPU brauchen, überprüfen wir, ob die aktuelle Hardware die CPU ist.
                For Each Sensor In Hardware.Sensors 'Jeden Sensor heraussuchen
                    If Sensor.SensorType = OpenHardwareMonitor.Hardware.SensorType.Temperature Then ' Da wir die Temperatur brauchen überprüfen wir, ob der aktuelle Sensor die Temperatur ausliest.
                        If Sensor.Index = Core Then 'Überprüfen, ob der aktuelle Index der gewünschte Kern ist
                            Return CDbl(Sensor.Value) 'Wenn alles passt (CPU, Temperatur-Sensor, Kern), wird der Wert zurückgegeben (MH/z)
                        End If
                    End If
                Next
            End If
        Next
        Return 0.0  'Wenn nichts gefunden wurde, wird '0.0" zurückgegeben.
    End Function

    Private Sub Timer1_Tick(sender As Object, e As EventArgs) Handles Timer1.Tick
     Computer.CPUEnabled = True
     Computer.Open()

'Hier wird der CPU Thread 0 aufgerufen um die Temperatur anzufragen.
     Label1.Text = "CPU Thread 1 " & "Temperatur: " & GetCoreTemp(0) & "°C"
End Sub
End Class
```

Naja deswegen hab ich auch dann nach weiteren Forschungen noch ein Link gefunden von einer fertigen v0.8.02 Alpha schon alles kompiliert.
[https://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.2-alpha.zip](url)
Da war auch die fertige exe dabei. Hab diese dann auch sofort getestet und das gleiche Ergebnis kam wie ich im ersten Post beschrieben habe.
Das selbe ist auch bei v0.8.0.3, v0.8.0.5.
Den Report von v0.8.0.5 hab ich ja bereits gepostet.
Die DLL wird von der bereits kompilierten Version bisher immer in Visual Basic angenommen und funktioniert auch, allerdings das selbe ergebnis wie bei der OpenhardwareMonitor.exe Datei die ich ausführe.

Das ende von Lied ist es erscheint auch wenn ich alle anderen 15 Threads probiere überall der Wert 0.

Das ist jetzt nur aus spass gemeint... 
>>> xDDD ich kann mir nicht vorstellen das der CPU ständig nur auf den Gefrierpunkt ist. <<<

Also jetzt ernsthaft wieder...
Darum wird warscheinlich bei der originalen exe Datei garnichts erst angezeigt, weil er probleme hat die Daten zu finden. 
richtig?

Ich merke gerade der Post ist etwas lange geworden, ich hoffe das stört aber keinen.

mit freundlichen grüssen,
Xiantrius

DarthDestroyer, 2018-08-18:
Hehe... Da hat jemand aber seinen Forschungsdrang ausgelebt ;)

Wenn du (Warum auch immer) nur die DLL compiliert bekommst, so kannst du die einfach in irgendeinen OHM Ordner rein kopieren und überschreiben! Die eigentliche "Hardwareerkennung" ist in der DLL und nicht in der Exe!

Die Version des Source-Codes ist "eigentlich" die aktuellste, der letzte Stand aller angenommen Pull Requests! Es gibt noch eine .3, die ist aber nicht wirklich weit hinterher und hilft vor allem bei den i7 etc.

Eigentlich scheinst du somit fast alles richtig gemacht zu haben! Probier es mal mit der DLL und dem Kopieren... Ansonsten gibt es glaube ich noch ein paar weitere Ryzen spezifische PRs ;)

DarthDestroyer, 2018-08-18:
Na siehst du, das klingt doch super! 

Ja, die Aqua und Libre Jungs haben das ganze etwas geändert und fortgeführt.
Leider hat man sich für die Framework Änderung entschieden, die ich persönlich auch erstmal als negativ deklarieren würde!
Weshalb bist du an 2.0 gebunden?

Wenn du jetzt super gut wärst, was GitHub angeht, dann könntest du all das jetzt als Pull Request für die Nachwelt festhalten ;) unter deinem Namen :D 

Freut mich aber, dass es läuft! Sehr schön!