openhardwaremonitor/openhardwaremonitor #1131: Crashes

xicemanx, 2018-09-13:
Hello,

I have been a big fan of Open Hardware Monitor, especially the gadget that you guys have.  However I find myself constantly moving it out of the way of a button.  One thing I did notice, and it has been between all of my computer is that the gadget stops responding when you try to move it at first.

Other than that please add support for the Nvidia Quadro K5100M

DarthDestroyer, 2018-10-26:
Dont know if the Gadget thing has something to do with OHM or Windows.
I cant reproduce it... 

For Nvidias industry Cards it looks dark! If they are not supporting the nvapi like the mining Cards, then there is no way, sadly

But as always, feel free to add a report for that!

xicemanx, 2018-10-26:
> Dont know if the Gadget thing has something to do with OHM or Windows.
> I cant reproduce it...
> 
> For Nvidias industry Cards it looks dark! If they are not supporting the nvapi like the mining Cards, then there is no way, sadly
> 
> But as always, feel free to add a report for that!

Hello DarthDestroyer,

Do you think it would be possible to make the gadget as a type of overlay that you can click through?  This would really make the transparency part of the gadget worth it in my opinion, and make it so you don't have to move it around much?  Just a thought.

DarthDestroyer, 2018-10-26:
There is a Sidebar Gadget App which use OHM as DataSource, i think it is able to do what you mean! 

https://github.com/ArcadeRenegade/SidebarDiagnostics