openhardwaremonitor/openhardwaremonitor #1135: No CPU-Temperature on Lenovo ThinkCentre m720t with Intel i3-8100 CPU

gunnarhaslinger, 2018-09-21:
Lenovo ThinkCentre m720t with Intel i3-8100 CPU
No CPU Temperature shown:
[image]

Here is the saved Report:
[OpenHardwareMonitor_Lenovo-ThinkCentre-m720t_Intel_i3-8100.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/2404648/OpenHardwareMonitor_Lenovo-ThinkCentre-m720t_Intel_i3-8100.Report.txt)

Would be very nice and helpful, if this System could be supported.
**Thank you, for your great Tool and Support!**

winner69, 2018-10-21:
http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

gunnarhaslinger, 2018-10-21:
Thanks, great!

closed 2021-08-30 by gunnarhaslinger