openhardwaremonitor/openhardwaremonitor #1139: i7 920 not showing clocks or temperature (0.8.0.3 alpha, report attached)

se1by, 2018-10-11:
As the title says, I can't see clocks or temperatures for the i7 920, it shows just a dash in the value and Max column. Given that this is a fairly old cpu (10 years!) I'm wondering if I'm doing something wrong?

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/2468231/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2018-10-11:
yep, you do =P

Ring0

Status: Installing driver "C:\Users\Jonas\Documents\openhardwaremonitor-v0.8.0.3-alpha\OpenHardwareMonitor\OpenHardwareMonitorLib.sys" failed and file exists
First Exception: OpenSCManager returned zero.
Second Exception: OpenSCManager returned zero.

Did you run it as admin?

se1by, 2018-10-14:
Running as admin shows clocks and temperature :+1: 
@DarthDestroyer Is there any way to have it ask for credentials at startup (using the "run at windows startup" option)?

Closing the issue as the original problem is solved.

closed 2018-10-14 by se1by