openhardwaremonitor/openhardwaremonitor #1141: Do not reduce memory on null sensor value

tralladas, 2018-10-19:
GetSensorValuesFromSettings method from Sensor class does not remove string value from memory when settings.GetValue(name, null) returns null.

Proposed change:

`private void GetSensorValuesFromSettings() {
      string name = new Identifier(Identifier, "values").ToString();
      string s = settings.GetValue(name, null);
            if (s == null)
            {
                settings.Remove(name);
                return;
            }
      try {
        byte[] array = Convert.FromBase64String(s);
        s = null;
        DateTime .......`

DarthDestroyer, 2018-10-26:
I would recommend to create a Pull Request out of that!
It is a small thing, but it would be better to track!

rmattuschka, 2018-11-02:
that is superfluous, since at the end of the function all "strings" are removed!

      // remove the value string from the settings to reduce memory usage
      settings.Remove(name);
}

tralladas, 2018-11-02:
Yes, you are right. The important line is not removing the setting but avoiding throwing the null reference exception, so the