openhardwaremonitor/openhardwaremonitor #1143: Support for Z390 with 9th gen Intel chips

Sulo-Hayha, 2018-11-11:
Using 0.8.0.3 Alpha

Missing:
- MOBO Temps
- CPU Clocks, Temps
- RAM Temp
- SSD Temp

Can't remember if this software ever did Volts and Watts but can't see them on anything.

DarthDestroyer, 2018-11-12:
First of all, a Report out of OHM would be helpful ;-)
Maybe it is already solved...

Sulo-Hayha, 2018-11-12:
[HWMonitor.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/2572386/HWMonitor.txt)

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/2572401/OpenHardwareMonitor.Report.txt)

Sulo-Hayha, 2018-11-12:
Moved it over to local drive and tried launching as admin.
That got all CPU related data to work, still missing the other 3.

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/2572601/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2018-11-12:
Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0xD42B at 0x2E/0x2F

Maybe, it is already solved... The Mainboard is Unknown in 0.8.3, but that is probably easy to fix ;-)

josask, 2018-12-14:
Please can you help me with this? 
[HWMonitor.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/2679992/HWMonitor.txt)

DarthDestroyer, 2018-12-14:
@josask, without a Report... Its hard to help, sorry (OHM Report)

BadlyAnticipated, 2018-12-22:
I'm missing my motherboard temps and vcore and my nvme drive info.

Hope somebody can fix it ;)

Here's my report:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/2705249/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2018-12-22:
And you are wrong in this Issue, too @BadlyAnticipated 

You use a 0x8688, too! (like josask)

More Details => 
Chip ID: Unknown ITE with ID 0x8688 at 0x2E/0x2F

Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0x8733 at 0x4E/0x4F

Chip ID: Unknown SMSC with ID 0x8733 at 0x4E/0x4F

BadlyAnticipated, 2018-12-22:
Thnx for the reply. Still ... I need help fixing it myself. I somehow managed to fix it showing my gpu memory values before ...was happy that worked. 

Will somebody ever create a new executable to download ?

DarthDestroyer, 2018-12-22:
it isnt really hard, as you already use the "compile it yourself"-version, you should be able to get that done! Most Chipsets are compatible from the past - only the link to the "new" one is missing.
Add the link and it should work!

If it works, add a Pull Request and help everyone else! ;)

BadlyAnticipated, 2018-12-25:
I tried ;) I could get the ITE8688 to show under my gigabyte z390 aorus ultra but it did not show anything below it sadly ... 

Would like some more help if you please ;)

DarthDestroyer, 2018-12-25:
The Question is, what did you change ;-)

Try out this  #1137  and the changed files

Merry Christmas

DarthDestroyer, 2018-12-25:
Could you attach a Report from your compiled Version?

Thanks mate! Enjoy the coding and debugging at Christmas ;)

BadlyAnticipated, 2018-12-25:
of course ;) thanx for being there bud ;)
incoming

BadlyAnticipated, 2018-12-25:
here is the report:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/2709031/OpenHardwareMonitor.Report.txt)

BadlyAnticipated, 2018-12-25:
loving the monitoring program btw ... have been using it for years ... like it above all others ... 

The gadget without a background is my thing ... omg ... do i prefer it above any other i found.

DarthDestroyer, 2018-12-25:
That looks sweet! Great Job!

I cant really test more, but i can recommend you this => #1061
There is some Info about 0x8733, too

Maybe that helps!

BadlyAnticipated, 2018-12-26:
I tried with a fresh clone of the repo. Changed all settings from #1061 
Still it won't show me my vcore and info. Sadly.

BOLL7708, 2019-01-20:
Got here through [this](https://github.com/openhardwaremonitor/openhardwaremonitor/issues/1148#issuecomment-441163281) comment.

With that alpha version I do get temperatures and core speeds, but the core speeds are static 🤔 even when running CPU benchmarks 😅 HWMonitor does display the clocks speeds. Reports attached below, which might be helpful. 

[HWMonitor.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/2776518/HWMonitor.txt)
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/2776519/OpenHardwareMonitor.Report.txt)

As a side note, I use the JSON server to get performance figures into my OBS overlay, very appreciated feature 😁