openhardwaremonitor/openhardwaremonitor #1144: Asus Z270-WS not supported

Slion, 2018-11-16:
See attached report.
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/2590420/OpenHardwareMonitor.Report.txt)

I that Nuvoton chip is not supported.

Slion, 2018-11-16:
I was running master GitHub branch.

Slion, 2018-11-16:
So I guess all I need is #884

DarthDestroyer, 2018-11-16:
Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0xD121 at 0x2E/0x2F

If that is listed there, probably yes ;-)

Slion, 2018-11-17:
@DarthDestroyer any idea where @moellerm is? That repository is in need of maintenance.

DarthDestroyer, 2018-11-17:
I totally agree and i have no idea!
I do my best to extinguish the flames but yes, we could need him! 

The best bet is, to wait until christmas / holiday time... 
But please, do not unpack the pitchfork!

DarthDestroyer, 2018-11-17:
There is already a fork, the libre thing...

Slion, 2018-11-17:
Where is it?

DarthDestroyer, 2018-11-17:
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor

Slion, 2018-11-17:
Sounds good, not sure what I'm doing here then...

DarthDestroyer, 2018-11-17:
there are some pros but also cons...
try it out ;-)