openhardwaremonitor/openhardwaremonitor #1145: H340 and i7-9700 issue

tabmag, 2018-11-17:
The i7-9700 only shows Load. Missing Temperatures, Bus Speed, and Powers.

tabmag, 2018-11-17:
I  restarted as admin, it still only shows Load for the i7.. I don't know what you mean by add another one. I don't se add as an option anywhere.

DarthDestroyer, 2018-11-17:
I meant the Report!
Without Admin, it will not work

DarthDestroyer, 2018-11-17:
And what i can say instantly => http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

tabmag, 2018-11-17:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/2592236/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2018-11-17:
Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0xD121 at 0x2E/0x2F

tabmag, 2018-11-17:
The v0.8.0.3 alpha works.

DarthDestroyer, 2018-11-17:
For the Mainboard => #981 should help, tested in #1060

tabmag, 2018-11-17:
Yeah, the v0.8.0.3 alpha works for the i7 issue, but not the Mainboard.

pkhdsth, 2018-11-18:
you can download https://www.cpuid.com/downloads/hwmonitor/hwmonitor_1.37.zip
run it, then upload report of hwmonitor, hope i can help you :D

DarthDestroyer, 2018-11-18:
@pkhdsth it is already solved ;-)

tabmag, 2018-11-18:
The CPUID HWMonitor displays the mother board information, Open Hardware Manager does not, but CPUID HWMonitor has no option to change the temperature to Farenheit or to run at startup. And Open Hardware Manager loads much faster, if only it would display the motherboard info.

tabmag, 2018-11-18:
Oh, CPUID does show Farenheit in parenthesis next to Celsius.