openhardwaremonitor/openhardwaremonitor #1149: ryzen 2700x erroneous CPUtemp figures

dav8208, 2018-11-26:
Hello, MB asrock x470 taichi. HWmonitor shows more accurate figures, but it couldnot be used with external temperature monitor.
Tried to find info about this problem in issues posts, but no success.
Sorry for my english

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/2616065/OpenHardwareMonitor.Report.txt)

[image]

rmattuschka, 2018-11-29:
You can take a look on Github's "Librehardwaremonitor" Repository.
There is more Ryzen - Support implemented. It is a Fork from here.

dav8208, 2018-11-29:
well, I found build where CPU temp is real but there is no menu to transmit info at external device which i use.
[image]

[image]

dav8208, 2018-11-29:
Thanks for support, but I found that an offset can be set for each parameter. (set -59 C for CPU, now looks real)