openhardwaremonitor/openhardwaremonitor #1153: Nvidia RTX 2080

Fyler74, 2018-12-06:
Fans of new RTX 2080 won't display.

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/2652556/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2018-12-06:
Cooler Settings

 Status: NOT_SUPPORTED

Maybe a Driver / Nvidia Issue, maybe unsupported Custom Design/PCB
That is not a OHM Issue

TommyGKiwi, 2018-12-28:
I have the same problem. Noticed it is working well using "HW monitor".

[image]

The second fan seems to work though.

DarthDestroyer, 2018-12-28:
@TommyGKiwi a Report would show more Details!

The Nvidia Class could need some Love, but lets see what the Report say!

TommyGKiwi, 2018-12-28:
here you go , ... :)
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/2715119/OpenHardwareMonitor.Report.txt)

TommyGKiwi, 2019-03-22:
debugged down to the NVAPI call .....
The call is ok, but says it is unsupported
checked with NVAPI docs and everything looks good except it uses only one fan (2080 I have has at least 2 tachs HWinfo64 has 2)
Nvidia came back to say currently the new cards are not properly supported in NVAPI.

twilingway, 2019-09-15:
anybody fixed it?