openhardwaremonitor/openhardwaremonitor #1160: No temps for i7-8750

paulqui, 2019-01-13:
OHM v 0.8.0 beta 

Temps for i7-8750 not being displayed

Please advise.

Thanks!

DarthDestroyer, 2019-01-13:
You have two Options!

http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

Please add a Report!

Without that, it is impossible to help you
By the way, it would be awesome, that you search before you write a new Issue, there are many similar Posts! Just for the next time ;)

Regards

paulqui, 2019-01-13:
Sorry I searched for "8750" and nothing came up.
Thanks for your reply.

DarthDestroyer, 2019-01-13:
"i7-8" would have brought you some result ;-)

aster94, 2019-01-20:
@paulqui did you tried the 8.0.3-alpha? was it solved?

I have the same cpu, this is my report, hoping this could help

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/2776939/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2019-01-20:
CPU should work with .3, yes