openhardwaremonitor/openhardwaremonitor #1166: Data missing on AMD FX-4100

felipebraz, 2019-02-04:
I can see all informations just fine on UI or report, but using the DLL on C# I´m missing:

- CPU Temperatures
- Fan Speed
- MB Temperature

DarthDestroyer, 2019-02-04:
How do you use it?

Could you provide the Part of your Code?

DarthDestroyer, 2019-02-17:
`Computer computer = new Computer();
computer.Open();
computer.MainboardEnabled = true;
computer.CPUEnabled = true;
computer.GPUEnabled = true;
computer.HDDEnabled = true;
computer.RAMEnabled = true;
computer.FanControllerEnabled = true;

Console.WriteLine("==========================");
foreach (IHardware hardware in computer.Hardware)
{
    Console.WriteLine("--------------------------");
    hardware.Update();
    Console.WriteLine(hardware.Name);
    Console.WriteLine(hardware.HardwareType);

    foreach (ISensor sensor in hardware.Sensors)
    {
        Console.WriteLine(sensor.Name);
        Console.WriteLine(sensor.Hardware);
        Console.WriteLine(sensor.SensorType);
    }
}`

Try that, with your lib you have! If that will fail, which lib do you use (from which version)?

felipebraz, 2019-02-19:
Same issue. I´ve tried two different ways:

1- Downloaded the openhardwaremonitor-v0.8.0-beta from the site (https://openhardwaremonitor.org/downloads/) and used the OpenHardwareMonitorLib.dll
2- Downloaded the full repository from here and imported the Lib to my project

In both cases, same result. The code:

Computer c = new Computer()
        {
            GPUEnabled = true,
            CPUEnabled = true,
            MainboardEnabled = true,
            HDDEnabled = true,
            RAMEnabled = true,
            FanControllerEnabled = true
        };

c.Open();

(...)

Console.WriteLine("==========================");
            foreach (IHardware hardware in c.Hardware)
            {
               Console.WriteLine("--------------------------");
                hardware.Update();
                Console.WriteLine("Hw: " + hardware.Name);
                Console.WriteLine("Hw Type: " + hardware.HardwareType);

                foreach (ISensor sensor in hardware.Sensors)
                {
                    Console.WriteLine("Sensor Name: " + sensor.Name);
                    Console.WriteLine("Sensor Hw: " + sensor.Hardware);
                    Console.WriteLine("Sensor Type: " + sensor.SensorType);
                }
            }

(...)

And the result:

==========================
--------------------------
Hw: Gigabyte GA-990FXA-UD3
Hw Type: Mainboard
--------------------------
Hw: AMD FX-4100
Hw Type: CPU
Sensor Name: CPU Core #1
Sensor Hw: OpenHardwareMonitor.Hardware.CPU.AMD10CPU
Sensor Type: Load
Sensor Name: CPU Core #2
Sensor Hw: OpenHardwareMonitor.Hardware.CPU.AMD10CPU
Sensor Type: Load
Sensor Name: CPU Core #3
Sensor Hw: OpenHardwareMonitor.Hardware.CPU.AMD10CPU
Sensor Type: Load
Sensor Name: CPU Core #4
Sensor Hw: OpenHardwareMonitor.Hardware.CPU.AMD10CPU
Sensor Type: Load
Sensor Name: CPU Total
Sensor Hw: OpenHardwareMonitor.Hardware.CPU.AMD10CPU
Sensor Type: Load
Sensor Name: CPU Core #1
Sensor Hw: OpenHardwareMonitor.Hardware.CPU.AMD10CPU
Sensor Type: Clock
Sensor Name: CPU Core #2
Sensor Hw: OpenHardwareMonitor.Hardware.CPU.AMD10CPU
Sensor Type: Clock
Sensor Name: CPU Core #3
Sensor Hw: OpenHardwareMonitor.Hardware.CPU.AMD10CPU
Sensor Type: Clock
Sensor Name: CPU Core #4
Sensor Hw: OpenHardwareMonitor.Hardware.CPU.AMD10CPU
Sensor Type: Clock
--------------------------
Hw: Generic Memory
Hw Type: RAM
Sensor Name: Memory
Sensor Hw: OpenHardwareMonitor.Hardware.RAM.GenericRAM
Sensor Type: Load
Sensor Name: Used Memory
Sensor Hw: OpenHardwareMonitor.Hardware.RAM.GenericRAM
Sensor Type: Data
Sensor Name: Available Memory
Sensor Hw: OpenHardwareMonitor.Hardware.RAM.GenericRAM
Sensor Type: Data
--------------------------
Hw: AMD Radeon HD 7800 Series
Hw Type: GpuAti
Sensor Name: GPU Core
Sensor Hw: OpenHardwareMonitor.Hardware.ATI.ATIGPU
Sensor Type: Temperature
Sensor Name: GPU Fan
Sensor Hw: OpenHardwareMonitor.Hardware.ATI.ATIGPU
Sensor Type: Fan
Sensor Name: GPU Fan
Sensor Hw: OpenHardwareMonitor.Hardware.ATI.ATIGPU
Sensor Type: Control
Sensor Name: GPU Core
Sensor Hw: OpenHardwareMonitor.Hardware.ATI.ATIGPU
Sensor Type: Clock
Sensor Name: GPU Memory
Sensor Hw: OpenHardwareMonitor.Hardware.ATI.ATIGPU
Sensor Type: Clock
Sensor Name: GPU Core
Sensor Hw: OpenHardwareMonitor.Hardware.ATI.ATIGPU
Sensor Type: Voltage
Sensor Name: GPU Core
Sensor Hw: OpenHardwareMonitor.Hardware.ATI.ATIGPU
Sensor Type: Load
--------------------------
Hw: Generic Hard Disk
Hw Type: HDD
Sensor Name: Used Space
Sensor Hw: OpenHardwareMonitor.Hardware.HDD.GenericHarddisk
Sensor Type: Load
--------------------------
Hw: Generic Hard Disk
Hw Type: HDD
Sensor Name: Used Space
Sensor Hw: OpenHardwareMonitor.Hardware.HDD.GenericHarddisk
Sensor Type: Load

DarthDestroyer, 2019-02-19:
What do you get from computer.GetReport() ?

I noticed a similar problem, the "fix" for that was something like:

If the Mainboard doesnt report any Value...
(...)
```
computer.Close()
computer.Open()
```
(...)

I havent found the "why", but the Mainboard is sometimes a little bit laggy and doesnt respond / exist in or at the first "run"

felipebraz, 2019-02-24:
Same thing... In sensors I just receive this:

--------------------------------------------------------------------------------

Sensors

|
+- Gigabyte GA-990FXA-UD3 (/mainboard)
|
+- AMD FX-4100 (/amdcpu/0)
|  +- CPU Core #1    :  3624.27  3624.27  3624.27 (/amdcpu/0/clock/1)
|  +- CPU Core #2    :  3624.27  3624.27  3624.27 (/amdcpu/0/clock/2)
|  +- CPU Core #3    :  3624.27  3624.27  3624.27 (/amdcpu/0/clock/3)
|  +- CPU Core #4    :  3624.27  3624.27  3624.27 (/amdcpu/0/clock/4)
|  +- CPU Total      :  66.6667  66.6667  66.6667 (/amdcpu/0/load/0)
|  +- CPU Core #1    :  83.3333  83.3333  83.3333 (/amdcpu/0/load/1)
|  +- CPU Core #2    :       50       50       50 (/amdcpu/0/load/2)
|  +- CPU Core #3    :  58.3333  58.3333  58.3333 (/amdcpu/0/load/3)
|  +- CPU Core #4    :       75       75       75 (/amdcpu/0/load/4)
|
+- Generic Memory (/ram)
|  +- Memory         :                            (/ram/load/0)
|  +- Used Memory    :                            (/ram/data/0)
|  +- Available Memory :                            (/ram/data/1)
|
+- AMD Radeon HD 7800 Series (/atigpu/0)
|  +- GPU Core       :    0.825    0.825    0.825 (/atigpu/0/voltage/0)
|  +- GPU Core       :      300      300      300 (/atigpu/0/clock/0)
|  +- GPU Memory     :      150      150      150 (/atigpu/0/clock/1)
|  +- GPU Core       :       37       37       37 (/atigpu/0/temperature/0)
|  +- GPU Core       :       41       41       41 (/atigpu/0/load/0)
|  +- GPU Fan        :     1101     1101     1101 (/atigpu/0/fan/0)
|  +- GPU Fan        :       20       20       20 (/atigpu/0/control/0)
|
+- Generic Hard Disk (/hdd/0)
|  +- Used Space     :                            (/hdd/0/load/0)
|
+- Generic Hard Disk (/hdd/1)
|  +- Used Space     :                            (/hdd/1/load/0)

--------------------------------------------------------------------------------