openhardwaremonitor/openhardwaremonitor #1167: Not understanding temperature readings via WMI

djloot, 2019-02-06:
I was finally able to get WMI working remotely
 (the array is returned with a space, to its referenced as [' Value'] instead of ['Value'])
 but don't understand readings.
The JSON response I get from WMI for temperature (example, Intel CPU0 Core #1 Temperature)
is as follows:
 1110441984.000000
 1110441984.000000
 1110441984.000000
 1110441984.000000
 1110704128.000000
 1110704128.000000
 1110704128.000000
 1110704128.000000
 1110704128.000000
 1110704128.000000

Can someone explain this to me? I had read somewhere that this number should be in millicelcius but that still doesn't make sense. Is it a delta from another number?

djloot, 2019-02-06:
Hoping someone can help me out here. 

The WMI results give me something like this:
        " Min": " 1110441984.000000", 
        " Index": " 3", 
        " Name": " CPU Core #4", 
        " Parent": " /intelcpu/0", 
        " InstanceId": " 3848", 
        " ProcessId": " d13850e6-8dbd-49be-95f3-b3e8a43a6a1a", 
        " Max": " 1114112000.000000", 
        " Value": " 1111228416.000000", 
        " SensorType": " Temperature", 
        "Identifier": "/intelcpu/0/temperature/3"

I don't know what to do with those numbers.  
The actual code that calculate it is:
if (packageTemperature != null) {
        uint eax, edx;
        // if reading is valid
        if (Ring0.RdmsrTx(IA32_PACKAGE_THERM_STATUS, out eax, out edx,
            1UL << cpuid[0][0].Thread) && (eax & 0x80000000) != 0) 
        {
          // get the dist from tjMax from bits 22:16
          float deltaT = ((eax & 0x007F0000) >> 16);
          float tjMax = packageTemperature.Parameters[0].Value;
          float tSlope = packageTemperature.Parameters[1].Value;
          packageTemperature.Value = tjMax - tSlope * deltaT;
        } else {
          packageTemperature.Value = null;
        }
      }

I see some bit AND code and some bit shifting going on but no clue how "eax" is even calculated. Can't find out anything online.

djloot, 2019-02-06:
Unfortunately after doing more research, the issue is that we need to get data from the processor itself about TJMax and other info, which is NOT provided via WMI. In essence, WMI for this product is useless, it gives you incomplete information. I'll see if I can scrape the info from the webserver instead. Frustrating

Artem-Mamzikov, 2020-01-28:
WMI requests stopped working, or rather, the answer came. How to restore / reinstall or fix this error?
[image]