openhardwaremonitor/openhardwaremonitor #1169: Incorrect temperature for AMD A4-4000 on ASUS A55BM-K

Tacis, 2019-02-17:
The environment has >20 degrees temperature. Other monitoring software show >60 degrees for CPU. Please fix this, I really need OHM to work with A4-4000. Thank you.
[image]

DarthDestroyer, 2019-02-17:
without a report, no one is able to help you...

Tacis, 2019-02-17:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/2873519/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2019-02-17:
First of all, outdated Version http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

Chip ID: Unknown ITE with ID 0x8603 at 0x2E/0x2F

Tacis, 2019-02-17:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/2873527/OpenHardwareMonitor.Report.txt)