openhardwaremonitor/openhardwaremonitor #1170: C# Can't get RAM Avaible and Used on OpenHardwareMonitorLibrary

pallinarozza, 2019-02-19:
I'm trying to get all avaible values on OpenHardwareMonitorLibrary using C#. I get all right values, but I can't get values reguarding RAM. 
I get only that as info:
Memory: MAX=
Used Memory: MAX=
Available Memory: MAX=

(all other values shows as this GPU Memory:1,497396 MAX=1,497396)
If I use the exe I can see properly the RAM Values. 
Is there something different between the exe and dll or I am missing something in my code?

This is my code
`
        static void GetSystemInfo()
        {
            Computer computer = new Computer();
            computer.Open();
            computer.CPUEnabled = true;
            computer.GPUEnabled = true;
            computer.RAMEnabled = true;

            for (int i = 0; i < computer.Hardware.Length; i++)
            {
               for (int j = 0; j < computer.Hardware[i].Sensors.Length; j++)
                    Console.WriteLine(computer.Hardware[i].Sensors[j].Name + ":" + computer.Hardware[i].Sensors[j].Value.ToString() + " MAX="+ computer.Hardware[i].Sensors[j].Max.ToString());               
            }     
        computer.Close();
        }`

pallinarozza, 2019-02-19:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/2877835/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2019-02-19:
Please Check this https://github.com/openhardwaremonitor/openhardwaremonitor/issues/1166#issuecomment-464439922

closed 2019-03-11 by lKinderBueno