openhardwaremonitor/openhardwaremonitor #1173: Missing temp for drive

antmck2016, 2019-03-20:
Just installed a new M.2 NVMe SSD drive on my Asus Maximus Hero 8 board.  I have 2 SATA drives as well.  Now I can only get temp for 2 of these drives - any and only 2.  If I change which connector the 2 SATA drives use then the temp changes also.  So depending which SATA connector I use on the motherboard I might not get a temp for the SSD drive or if I change SATA connectors I get a temp for that drive but not one of the other SATAs.

DarthDestroyer, 2019-03-20:
#984 could help here?

without a Report, there is nothing more possible to say... could you attach one?

3c71, 2020-04-29:
Got 2 NVMe that are listed as 'Generic Hard Disk', one is a INTEL SSDPEKNW010T8, the other a Force MP500.

I've attached the report in case it helps.
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/4552256/OpenHardwareMonitor.Report.txt)

3c71, 2020-05-16:
Here is some information about some Intel NVMe drive that is not recognized as SSD. I've added some debugging in the detection and logged all attributes found on each drive with smart enabled. My Drive #4 is INTEL SSDPEKNW010T8, but read NIET LSSPDKEWN10T0 8. 

Also the list of attributes seem to vary from one run to the next:

Smart enabled for 4
Attribute 0xD6 - System.Byte[]
Attribute 0xAE - System.Byte[]
Attribute 0xFF - System.Byte[]
Attribute 0xB1 - System.Byte[]
Attribute 0xFF - System.Byte[]
Attribute 0xF1 - System.Byte[]
Attribute 0xB1 - System.Byte[]
Attribute 0xAF - System.Byte[]
Attribute 0x4A - System.Byte[]
Attribute 0xBC - System.Byte[]
Attribute 0xFF - System.Byte[]
Found SMART drive NIET LSSPDKEWN10T0 8 for 00C2

Smart enabled for 4
Attribute 0x30 - System.Byte[]
Attribute 0xFF - System.Byte[]
Attribute 0x40 - System.Byte[]
Attribute 0xFF - System.Byte[]
Attribute 0x70 - System.Byte[]
Attribute 0x6 - System.Byte[]
Attribute 0xFF - System.Byte[]
Attribute 0x14 - System.Byte[]
Attribute 0xFF - System.Byte[]
Attribute 0x70 - System.Byte[]
Attribute 0xFF - System.Byte[]
Attribute 0xFF - System.Byte[]
Attribute 0xFF - System.Byte[]
Attribute 0xFE - System.Byte[]
Attribute 0x30 - System.Byte[]
Attribute 0xF9 - System.Byte[]
Found SMART drive NIET LSSPDKEWN10T0 8 for 00C2