openhardwaremonitor/openhardwaremonitor #1174: No clocks or temp for intel core i7 6820HQ

sirKitKat, 2019-03-21:
Screenshot:
[image]

Report:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/2992633/OpenHardwareMonitor.Report.txt)

Thanks!

DarthDestroyer, 2019-03-21:
Download the latest Version and run as Admin ;-)

http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

sirKitKat, 2019-03-21:
Thx, I was running the latest version (0.8.0 beta) and running that version as admin did the trick.
Suggestion: when hovering over the empty values and you are not running the app as admin, maybe a tool-tip could appear with the suggestion to run the program as admin to view the missing values?

closed 2019-03-21 by sirKitKat