openhardwaremonitor/openhardwaremonitor #1175: The use of Openhardwaremonitor.dll

Breno94re, 2019-04-02:
Hey guys, i'm devloping a project for my computer science college. The idea is to monitoring servers via network wih C#.  Yours library shed me alot of Light, and i'm here to ask if i can use it for this non-comercial project, only with academic porpuses. It helped me to understand how mwi works and i'll give credit to your work.  Thanks in advance!

DarthDestroyer, 2019-04-02:
I am not a lawyer, but OHM is open source and at that point i dont see any problems with a non-commercial use!
I dont know the specific license, but it should be enough to check that out...
and with credits on top, you should be fine! 

Feel free to give us more Informations of your Project! Sounds great! 

I wish you success!

Breno94re, 2019-04-02:
Thanks on the quick response! I figured that was ok to use for non-comercial porpuses, but asking doesnt hurt hehe. Besides, i was trying to solve the problem on ryzen readings, getting through on alot of issues, i saw one of the devs saying that alot of projects are using the library without even asking if they could do it. So,  i wanted to do it the right way. 

Well,  hope you guys keep the good progress ongoing! Thanks again!

nobodysu, 2019-04-07:
Academic purposes are great!
But you, or some other person might not want to reinvent the wheel - there are already a couple solutions, such as:
https://github.com/nobodysu/zabbix-mini-IPMI
This one is built on top of Zabbix. That's self-promotion, yeah.

Breno94re, 2019-04-07:
I dont think this as reinvent the wheel,  most of programs out there with this porpuse have limitations, my goal is to create something that any user can benefit from it,  with a nice design and the option to have total control on your pc from distance. Not just shutting it down, but the option to do reverse remote control, like Team viewer, without the need to do anything besides clicking a button. Porting it to android on the future its a goal too. The main goal is to make it open source and help others to learn too,  the code have commentarys on every corner,  to make sure people will learn something from it.  Basically, thats it. But i do appreciate the advise!