openhardwaremonitor/openhardwaremonitor #1176: Update of Open Hardware Monitor

Eostel, 2019-04-03:
Hello,

I am a user of Open Hardware Monitor since many years now (nearly 5). I love the possibility to make curves and watch temperatures, power and usage evolution.

I recently buy a Coffe lake Refresh CPU with a Z390 MSI motherboard and now, I can't use Open Hardware Monitor for Viewing temperatures or Power. I just can have a view on the usage in % of the CPU. CPUID shows me temperature and power but i really prefer Open Hardware monitor for the curves! <3

So my question is: Is any updates/ upgrades of Open hardware Monitor planned to fully incorporate Coffe lake Refresh/Z390 monitoring?

Thank you! Best regards

Eostel

DarthDestroyer, 2019-04-03:
First of all, it is nearly impossible to help you without a Report! 

After that, there are a lot of other Issues (with a Solution) with the same Topic! 
Have you checked them?

jyxjjj, 2019-08-08:
So am i !
Motherboard: ASUS PRIME Z390-P
CPU: Intel Core I7-9700K
GPU: ASUS NVIDIA RTX 2060

[DxDiag.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/3479374/DxDiag.txt)

Don't know which "Report" you need. I may provide.

jyxjjj, 2019-08-08:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/3479385/OpenHardwareMonitor.Report.txt)
I found it.

jyxjjj, 2019-08-08:
Searched in issue.
I found v0.8.0.3 & it fixed my problem.