openhardwaremonitor/openhardwaremonitor #1179: Read CPU Package Power with OpenHardwareMonitorLib

Crusoe86, 2019-04-15:
Hi Michael,

I have a little question:
I am writing a little Tool that reads some values of the computer and redirect it to an arduino 
For e.g. CPU temperature, CPU Load... But... I can't get the value for the CPU Package power.
it only shows the Power consumption for the CPU DRAM.
I'm new to C#, so I can't help myself... Even if I was studying your source code I didn't get it :(

Here are the parts of the code where I get my values:
```
[...]
       Computer c = new Computer()
        {
            GPUEnabled = true,
            CPUEnabled = true,
            RAMEnabled = true
        };
[...]
        public Form1()
        {
            InitializeComponent();
            Init();
            c.Open();
        }
[...]
private void Status()
        {
            foreach (var hardware in c.Hardware)
            {

               if (hardware.HardwareType == HardwareType.CPU)
                {
                    hardware.Update();
                    foreach (var sensor in hardware.Sensors)
                        if (sensor.SensorType == SensorType.Temperature)
                        {
                            value1 = sensor.Value.GetValueOrDefault(); // CPU Package Temp

                        }
                    foreach (var sensor in hardware.Sensors)
                        if (sensor.SensorType == SensorType.Load)
                        {
                            value2 = sensor.Value.GetValueOrDefault(); // CPU Total Load

                        }
                    foreach (var sensor in hardware.Sensors)
                        if (sensor.SensorType == SensorType.Power)
                        {
                            
                            value3 = sensor.Value.GetValueOrDefault(); // CPU Package Power (but shows DRAM Power)

                        }
                }

                if (hardware.HardwareType == HardwareType.GpuNvidia)
                {
                    hardware.Update();
                    foreach (var sensor in hardware.Sensors)
                        if (sensor.SensorType == SensorType.Temperature)
                        {

                            value4 = sensor.Value.GetValueOrDefault(); // GPU Temperature
                        }

                }

            }
```

I hope you can help me with this issue...
If you need the whole code, of course I will post it. But at first I won't spam the whole code to this post :)

Many regards
Crusoe

DarthDestroyer, 2019-04-15:
Hey,

i give you a small tipp and you will find the Problem ;-)

Dont put the values into one var! 
write them out to the console!

Crusoe86, 2019-04-15:
Hi Michael,

thank you for the quick response :)
I'm not sure if we talk at cross purposes^^

My "values 1-4" will be send to a COM Port:
`port.Write(value1 + "a" + value2 + "b" + value3 + "c" + value4 + "d");`

The letters a-d a some kind of seperators for the ardiuno grab the right value for the right field on the display.

I'm not sure if I understand your hint... Do you mean something like this?
`Console.Write(value3)`

DarthDestroyer, 2019-04-15:
Nono... I am not Michael ;-)

It is close to that, what i've tried to say... 

Dont use

value1 = sensor.Value.GetValueOrDefault(); 

use this one insted

console.write( "Test" + sensor.Value.GetValueOrDefault());

And you will quickly see the Problem ;-)

(Small hint: There should be more as 4 outputs to your console)

Crusoe86, 2019-04-15:
AHHHH Sorry, Dan >_<'

The Console.write does not do the trick...
But
`System.Diagnostics.Debug.WriteLine(value3);`
does 💃 
Debug Output:
```
32,31705
21,1089
0
1,345534
```

There are my values :D

Now I have to find out, how to grab the first value. This is the one I want :)

DarthDestroyer, 2019-04-15:
You are still wrong, mate!

Throw these "values1-4" away (atleast for the testing) ;-)

in every if-statement where you set these values with the sensor data, use the console output ;-)

Crusoe86, 2019-04-15:
I'll give that a shot... :)
But I think I will get more than I want.
I will try that.
JFYI: "value 1", "value 2" and "value 4" give the right output. Only "value 3" does not.
But good to know, that they a there :)

I will rewrite the code like you recommend and take a look at the Console (or Debug Window, did not find the Console).
Sorry. I am using Visual Studio 2019 for two days.
Normally I am working with AutoIt XD

Crusoe86, 2019-04-15:
As expected, I get way to much information :D
This is the Debug Windows Output:
```
value1: 51
value1: 46
value1: 44
value1: 42
value1: 44
value1: 41
value1: 51
value2: 21,875
value2: 1,5625
value2: 12,5
value2: 12,5
value2: 9,375
value2: 0
value2: 9,635417
value3: 43,52773
value3: 32,23359
value3: 0
value3: 1,758796
value4: 40
```

With this code:
```
               if (hardware.HardwareType == HardwareType.CPU)
                {
                    hardware.Update();
                    foreach (var sensor in hardware.Sensors)
                        if (sensor.SensorType == SensorType.Temperature)
                        {
                            value1 = sensor.Value.GetValueOrDefault();
                            System.Diagnostics.Debug.WriteLine("value1: " + sensor.Value.GetValueOrDefault());

                        }
                    foreach (var sensor in hardware.Sensors)
                        if (sensor.SensorType == SensorType.Load)
                        {
                            value2 = sensor.Value.GetValueOrDefault();
                            System.Diagnostics.Debug.WriteLine("value2: " + sensor.Value.GetValueOrDefault());

                        }
                    foreach (var sensor in hardware.Sensors)
                        if (sensor.SensorType == SensorType.Power)
                        {
                            
                            value3 = sensor.Value.GetValueOrDefault();
                            System.Diagnostics.Debug.WriteLine("value3: " + sensor.Value.GetValueOrDefault());

                        }
                }
```

I don't remove the "value1 = sensor.Value.GetValueOrDefault();" lines, cause the were not disturbing :)

Where do I get the Output if I use "Console.write" instead of "System.Diagnostics.Debug.WriteLine"?

DarthDestroyer, 2019-04-15:
I am on mobile at the moment, so i am limited to explaining ;-)

You could open OHM and make a Report out of it (File > Report) 
When you save and open it, you will see all possible values and sensors! 

At that Point, you will see what i try to explain... 
For example "load", there should be that much sensors as your cpu has real cores plus 1 for the overall sensor... 

followed by temps, every core is listed there with his own temps and the overall package... 

if you loop through all of them, only the last value will be set to the variable... Sadly, it is only visible at your power Statement! 
But it is not the last Power sensor you want ;-)

if you look into your Report, you will notice a longer identifier with a number, use that... or take another if statement with the sensor name to get the package Power or whatever you like ;-)

Crusoe86, 2019-04-15:
Thank you for your help so far :)
But there are definitly some misunderstandings XD

The Values are correct.
I can see, that there are more values behind my every variables "value1"-"value4".
I understand this.
The question is: How do I grab the right value out of the multiple entries behind "GetValueOrDefault()"?

DarthDestroyer, 2019-04-15:
```
if (hardware.HardwareType == HardwareType.CPU)
                {
                    hardware.Update();
                    foreach (var sensor in hardware.Sensors)
                        if (sensor.SensorType == SensorType.Temperature && sensor.Name.ToString.Contains("Package"))
                        {
                            value1 = sensor.Value.GetValueOrDefault();
                            System.Diagnostics.Debug.WriteLine("value1: " + sensor.Value.GetValueOrDefault());

                        }
                    foreach (var sensor in hardware.Sensors)
                        if (sensor.SensorType == SensorType.Load && sensor.Name.ToString.Contains("Total"))
                        {
                            value2 = sensor.Value.GetValueOrDefault();
                            System.Diagnostics.Debug.WriteLine("value2: " + sensor.Value.GetValueOrDefault());

                        }
                    foreach (var sensor in hardware.Sensors)
                        if (sensor.SensorType == SensorType.Power && sensor.Name.ToString.Contains("Package"))
                        {
                            
                            value3 = sensor.Value.GetValueOrDefault();
                            System.Diagnostics.Debug.WriteLine("value3: " + sensor.Value.GetValueOrDefault());

                        }
                }
```

That should do the trick ;)
Sorry for the long wait, had to go to the pc...

Crusoe86, 2019-04-15:
You are a fox!!!
But:
`if (sensor.SensorType == SensorType.Power && sensor.Name.ToString.Contains("Package"))`
does not work. 
It gives the following error:
"Fehler	CS0119	'"string.ToString()" ist "Methode" und im angegebenen Kontext nicht gültig."
I had to change the code to:
`if (sensor.SensorType == SensorType.Power && sensor.Name.Contains("Package"))`
This gives the value I want :)

Is it possible to read out the Sensor Names?
Where do you get this Name?
In the Report I would use this line:
`|  +- CPU Package    :  23.9628  7.47977  63.4548 (/intelcpu/0/power/0)`
And extract "CPU Package" as name.
Is it possible to use an argument for this function "GetValueOrDefault()" that holds the whole values in my if clause? Something that I can put as an option in the bracket?
In AutoIt it is possible to give some options in the brackets to control the function...
Did you know what I mean?

DarthDestroyer, 2019-04-15:
The Report is doing it like

sensor.Name + ":" + sensor.Value + " " + sensor.Min + " " + sensor.Max + "(" + sensor.Identifier + ")"

Of course you can give Parameters or Variables to functions, if they could use them...
"GetValueOrDefault()" means only, that it will always give a Value - doesnt matter if it fails or works!

You can see, that you will get the same result when you use "sensor.Value" or .Min / .Max

PS: Mit der Fehlermeldung auf Deutsch nehme ich an, dass Deutsch vielleicht einfacher wäre? :P Dann müssen wir nicht beide übersetzen?

Crusoe86, 2019-04-15:
Hell yeah... That's it 👯‍♂️ 
Now all is working as expected 👍 
Thank you again for your help 🥇 

Do you have a PayPal Account? 
I owe you one beer or two... ✌️ 
Without you I won't find a solution that fast and learned that much in this short time 👍 
C# noob since two days 😋

P.S. Das könnten wir zwar, aber dann würden andere mit einem ähnlichen Problem nichts verstehen XD

DarthDestroyer, 2019-04-15:
i just changed a little bit more of that code (edited the post above)

you are absolutly welcome, well done! 
i enjoy to help wherever i can, you owe me nothing mate! 
everyone is starting small 😉
switch your experience with mine and i bet my arduino knowledge is even under your c# skill 😅 
but the time will come, someday 🤣

but what you could do for me (or better us/everyone), publish your work! i would really like to see the result! 

PS: Freut mich geholfen zu haben! 
Dachte nur kurz, da waren sicher ein paar Missverständnisse in unserer Konversation, wo wir es auch einfacher hätten haben können 😁 

best regards