openhardwaremonitor/openhardwaremonitor #1180: No CPU Fan speed for Asus Zenbook UX410UQK

UserCel, 2019-04-15:
[OpenHardwareMonitor.Report-UX410UQK.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/3081744/OpenHardwareMonitor.Report-UX410UQK.txt)

There's a single fan for the cpu+gpu. Doesn't seem to show in the monitor.
Would appreciate it if there's a possible solution.

Thanks!

DarthDestroyer, 2019-04-15:
First of all, i would recommend the latest Version http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

But it will not solve your Problem with the Fan!
As i see, you use a Laptop... 
Yes, these things are not really supported because they dont use a "known" chipset to control the fan(-s).
Some of the Laptops use one of the known Chipsets and are supported, but every manufacturer is brewing his own soup =/ sadly!

UserCel, 2019-04-15:
I figured, that's too bad.
Thank you for your time anyways ( ´ ▽ ` )ﾉ

ebricca, 2019-06-28:
as i understand fan speed for laptops could be read out of the acpi registry (dsdt) though it's not for every model in the same place

nbfc (notebook fancontrol) could be worth a try to understand auto hinting / get some of the offsets to read the values

https://github.com/hirschmann/nbfc/wiki/Analyze-your-notebook's-DSDT  
https://github.com/hirschmann/nbfc/issues/246  
https://www.tonymacx86.com/threads/new-fan-control-dsdt-for-silent-fan-at-higher-temps.72043/

as for gpu temperature i didn't look into it .. what i found is that HWiNFO.exe is capable to read both infos in my case of a T530 - from the EC (embedded controller) as well as ACPI - could be used to check that the offsets are correct

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/3339813/OpenHardwareMonitor.Report.txt)