openhardwaremonitor/openhardwaremonitor #1182: Missing all stats for X370 Gaming Pro Carbon and temp/voltage for R3 2200G

rivermont, 2019-05-03:
[Partial report](https://pastebin.com/kgfzcTKu)  
I have no data reported for by MSI X370 Gaming Pro Carbon motherboard.  
The Ryzen stuff seems to be taken in #957

DarthDestroyer, 2019-05-04:
Your Version is not the latest, but you should find it in some last Issues as a link!

Could you upload the full report to this issue? The important informations are cut off!

rivermont, 2019-07-08:
This slipped past me!

Here is the full report:

```

Open Hardware Monitor Report

--------------------------------------------------------------------------------

Version: 0.8.0.0

--------------------------------------------------------------------------------

Common Language Runtime: 4.0.30319.42000
Operating System: Microsoft Windows NT 10.0.17763.0
Process Type: 64-Bit

--------------------------------------------------------------------------------

Sensors

|
+- X370 GAMING PRO CARBON (MS-7A32) (/mainboard)
|
+- AMD Ryzen 3 2200G with Radeon Vega Graphics (/amdcpu/0)
|  +- CPU Total      :  5.46875  4.29688  53.9063 (/amdcpu/0/load/0)
|  +- CPU Core #1    :  10.9375   4.6875  54.6875 (/amdcpu/0/load/1)
|  +- CPU Core #2    :    3.125    3.125       50 (/amdcpu/0/load/2)
|  +- CPU Core #3    :    3.125        0   59.375 (/amdcpu/0/load/3)
|  +- CPU Core #4    :   4.6875        0  54.6875 (/amdcpu/0/load/4)
|
+- Generic Memory (/ram)
|  +- Memory         :  34.2097   33.957  34.3096 (/ram/load/0)
|  +- Used Memory    :  5.45446  5.41417   5.4704 (/ram/data/0)
|  +- Available Memory :  10.4898  10.4738    10.53 (/ram/data/1)
|
+- Radeon RX 580 Series (/atigpu/0)
|  +- GPU Core       :      0.8      0.8      0.8 (/atigpu/0/voltage/0)
|  +- GPU Core       :      300      300   304.86 (/atigpu/0/clock/0)
|  +- GPU Memory     :      300      300      300 (/atigpu/0/clock/1)
|  +- GPU Core       :       45       45       45 (/atigpu/0/temperature/0)
|  +- GPU Core       :        0        0       38 (/atigpu/0/load/0)
|  +- GPU Fan        :        0        0        0 (/atigpu/0/fan/0)
|  +- GPU Fan        :        0        0        0 (/atigpu/0/control/0)
|
+- ST3250310AS (/hdd/0)
|  +- Temperature    :       42       41       42 (/hdd/0/temperature/0)
|  +- Used Space     :  80.4199  80.4195  80.4201 (/hdd/0/load/0)
|
+- TEAML5Lite3D120G (/hdd/1)
|  +- Temperature    :        0        0        0 (/hdd/1/temperature/0)
|
+- TEAML5Lite3D120G (/hdd/2)
|  +- Temperature    :        0        0        0 (/hdd/2/temperature/0)
|
+- WDC WDS250G2B0B-00YS70 (/hdd/3)
|  +- Temperature    :       42       42       42 (/hdd/3/temperature/0)
|  +- Used Space     :  16.5258  16.5258  16.5258 (/hdd/3/load/0)

--------------------------------------------------------------------------------

Parameters

|
+- X370 GAMING PRO CARBON (MS-7A32) (/mainboard)
|
+- AMD Ryzen 3 2200G with Radeon Vega Graphics (/amdcpu/0)
|
+- Generic Memory (/ram)
[... 684 lines cut ...]
 AA Unknown                            000000000000 100   100   -     -       
 AB Unknown                            000000000000 100   100   -     -       
 AC Unknown                            000000000000 100   100   -     -       
 AD Unknown                            020000000000 100   100   -     -       
 AE Unknown                            990000000000 100   100   -     -       
 B8 End-to-End error                   000000000000 100   100   -     -       
 BB Reported Uncorrectable Errors      000000000000 100   100   -     0       
 BC Command Timeout                    000000000000 100   100   -     0       
 C2 Temperature                        2A0001003E00 62    58    -     42      
 C7 UltraDMA CRC Error Count           000000000000 100   100   -     -       
 E6 GMR Head Amplitude                 220014002200 100   100   -     -       
 E8 Endurance Remaining                640000000000 100   100   4     -       
 E9 Power-On Hours (POH)               550200000000 100   100   -     -       
 EA Unknown                            200400000000 100   100   -     -       
 F1 Total LBAs Written                 650300000000 253   253   -     -       
 F2 Total LBAs Read                    B90200000000 253   253   -     -       
 F4 Unknown                            000000000000 100   0     -     -       

Logical drive name: D:\
Format: NTFS
Total size: 67108859904
Total free space: 56018579456
```

I can run it again on a newer version if needed.