openhardwaremonitor/openhardwaremonitor #1185: No CPU temp on "metallic"(bronze,silver,gold,etc)-class Xeons

Astara, 2019-05-28:
Has there been any release since Beta 8.0.3?  That's the last I tried (with no success).

Tried to fix it myself, but I didn't have the right MS-compiler+headers, so it wouldn't compile.

Useless report included:

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/3226092/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2019-05-28:
MS Visual Studio :-P

There is no other stable/beta/alpha release available!

Astara, 2019-05-28:
I was afraid of that....sigh

DarthDestroyer, 2019-05-28:
The best path would be:

Check the PRs, add what you need, give feedback or in the best case, create your own PR!

You can imagine, that it isnt possible to get all cpus/gpus and test them... With a little bit of work, you could make the difference ;-)