openhardwaremonitor/openhardwaremonitor #1187: Radeon VII no data

dnikonov, 2019-06-15:
Hey.
OHM does not give out data on the AMD Radeon VII card.
What data do you need to send?

[image]

DarthDestroyer, 2019-06-15:
You could add a Report which provide a little bit more info...

Attach the textfile to that issue ;-)

dnikonov, 2019-06-15:
Attached the report.

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/3293104/OpenHardwareMonitor.Report.txt)

dnikonov, 2019-06-24:
Hey.
Any news on the issue?

binary100100, 2019-09-20:
I'm having the exact same issue with RX 5700 cards. Perhaps it's because Navi?

moellerm, 2020-02-28:
Please try with this test version
http://openhardwaremonitor.org/files/openhardwaremonitor-v0.9.1.1-alpha.zip
and submit a report by selecting File -> Submit Report... from the menu.

dnikonov, 2020-02-29:
RTX 2060 - everything is fine
Radeon VII - fan speed in RPM, please add as a percentage, like other cards

[RTX 2060.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/4270565/RTX.2060.txt)
[RTX 2060 2.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/4270564/RTX.2060.2.txt)
[Radeon VII.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/4270566/Radeon.VII.txt)

moellerm, 2020-03-01:
Thank you for your feedback. I have added some additional sensors to the OHM release version 0.9.2. Fan speed as percentage can be read by OHM, but seems not to be provided by the GPU driver in your case.

closed 2020-03-01 by moellerm