openhardwaremonitor/openhardwaremonitor #1189: ASUS A88X-PLUS fan (Fintek F71811/812) solved

LazizEx, 2019-07-07:
I found how to determine the temperature and fan speed.
1: (Chips.cs)
`
internal enum Chip : ushort
    {
     ...
    F71811 = 0x1007,
    ...
    }
`

2: (LPCIO.cs)
`
private bool DetectWinbondFintek(LPCPort port)
        {

           port.WinbondNuvotonFintekEnter();
            byte logicalDeviceNumber = 0;
            byte id = port.ReadByte(CHIP_ID_REGISTER);
            byte revision = port.ReadByte(CHIP_REVISION_REGISTER);
            Chip chip = Chip.Unknown;
            switch (id)
            {
            ...
            case 0x11:
                    switch (revision)
                    {
                        case 0x18:
                            chip = Chip.F71811;
                            logicalDeviceNumber = FINTEK_HARDWARE_MONITOR_LDN;
                            break;
                    }
                    break;
`

Fintek F71811/812 compatible with F71869A and F71868AD
[image]

DarthDestroyer, 2019-08-02:
It would be awesome, if that would be available as a Pull Request 😉

LazizEx, 2019-08-26:
> 
> 
> It would be awesome, if that would be available as a Pull Request 😉

Ready)

spencydot, 2021-06-27:
what is

PrplHaz4, 2023-09-10:
PR is here and unmerged for anyone looking...but it seems support may have been added in other commits.

 https://github.com/openhardwaremonitor/openhardwaremonitor/pull/1197