openhardwaremonitor/openhardwaremonitor #1191: OpenHardwareMonitor installer / updater

mouuff, 2019-07-12:
Hello, we are a team of students (5th year in master computer engineering school in France)
We are currently working on a easy to setup installer/updater, which could be used for openhardwaremonitor.
The project is named Quail, it is hosted here: https://github.com/QuailTeam/iQuail
Currently the project is in development stage, and we are looking for projects like yours, to use our solution, and get feedbacks from your community and your team.
We started this project a year ago. Today we believe we have came to a point where it could be released in beta version.

Here is a screenshot of the user interface:
[image]

The configuration required to build an installer is that easy:

```
#!/usr/bin/python3

import iquail

iquail.run(
    solution=iquail.SolutionPacked(path='OpenHardwareMonitor'),
    installer=iquail.Installer(
        publisher="Michael Moller",
        name='OpenHardwareMonitor',
        icon='OpenHardwareMonitor.exe',
        binary='OpenHardwareMonitor.exe',
        console=True,
        launch_with_quail=False
    ),
    builder=iquail.builder.Builder(
        iquail.builder.CmdIcon('icon.ico'),
        iquail.builder.CmdNoconsole()
    ),
    controller=iquail.ControllerTkinter()
)
```

### How to try it?
Here is tutorial to install quail:
https://github.com/QuailTeam/iQuail/wiki/Windows-install

Here is a zip containing:
- a generated installer for openhardwaremonitor (unsigned binary)
- an example script and required files to generate an installer (generate it using `openhardwaremonitor.py --iquail_build`

### Automatic updates?
It is also possible to make self updating software using quail (using github release system, or by setting up a server).
The only thing which would be edited in the installation script is `iquail.SolutionPacked` by a distant "solution" (such as `iquail.SolutionGithub`)
Download examples [here](https://github.com/QuailTeam/iQuail/tree/master/examples) .

jyxjjj, 2019-10-14:
Passing this issue.
I don't know how others think, But i think the installer of all the programs are loathsome.
They will use system directory like ProgramData, temp, and so on.
They will make a pollution to regedit.
And most of that 
They may make a program don't use the installation directory to save files.
Not green, hate enough.

mouuff, 2019-10-14:
@jyxjjj 
"installer of all the programs are loathsome" Basically every applications on windows use an installer, most users don't want to unzip, copy, and create shortcut by themselves every-time there is an update.
"They will make a pollution to regedit" The windows registry is made for that purpose, when you open a website on firefox is it polluting your history? This is ridiculous
"They may make a program don't use the installation directory to save files" Thats just plain false.

jyxjjj, 2019-11-26:
> @jyxjjj
> “所有程序的安装程序都很讨厌”基本上Windows上的每个应用程序都使用安装程序，大多数用户不想每次进行更新时都自行解压缩，复制和创建快捷方式。
> “它们会对注册表造成污染” Windows注册表就是为此目的而创建的，当您在firefox上打开网站时，是否污染了您的历史记录？这太荒谬了：
> “他们可能使程序不使用安装目录来保存文件”。

Really? Explain why we have docker or k8s?
Of cause i always think not portable equals to dirty.

mouuff, 2019-11-26:
Honestly I don't understand the comparison with docker or k8s.
I also don't understand how it is not "portable".
It is is as portable as docker: this works on every operating system (unlike many many installers)
Plus something not portable doesn't make it dirty. (Which again is not the case with our project)