openhardwaremonitor/openhardwaremonitor #1192: CPU Temperatures Not Able to be Fetched:

Nova0ne, 2019-07-23:
Hello.

I have been going at an implementation of this library for a few days now, and I just cannot seem to get it to work.

Firstly, my CPU doesn't seem to want to return a temperature when I request it. Example below:
```
String temperature;
Computer localDevice = new Computer();
localDevice.Open();
localDevice.CPUEnabled = true;

foreach(var hardwareItem in localDevice.Hardware) {
	if (hardwareItem.HardwareType == HardwareType.CPU) {
		foreach(var sensor in hardwareItem.Sensors) {
			if (sensor.SensorType == SensorType.Temperature) {
				temperature = sensor.Name + ":" + sensor.Value.ToString();
			}
		}
	}
}
```

I have read that you "have to go through subhardware too," but when I do that it won't even parse subHardware.Hardware as if there is nothing to parse in the array at all!

On another note, when I am detecting the HardwareType as anything other than the CPU, it seems that it won't even pick up anything other than HardwareType.CPU - it won't even detect my NVIDIA GPU in GpuNvidia!

I really need this library for a hardware monitoring application I am creating, so any help will be very appreciated.

Thanks!

DarthDestroyer, 2019-07-31:
Take a look at https://github.com/openhardwaremonitor/openhardwaremonitor/issues/1166#issuecomment-464439922

Nova0ne, 2019-08-02:
> 
> 
> Take a look at [#1166 (comment)](https://github.com/openhardwaremonitor/openhardwaremonitor/issues/1166#issuecomment-464439922)

This is what we get. Any ideas?

````
=================
-------------------
Intel Core i7-8700K
CPU
CPU Core #1
OpenHardwareMonitor.Hardware.CPU.IntelCPU
Load
CPU Core #2
OpenHardwareMonitor.Hardware.CPU.IntelCPU
Load
CPU Core #3
OpenHardwareMonitor.Hardware.CPU.IntelCPU
Load
CPU Core #4
OpenHardwareMonitor.Hardware.CPU.IntelCPU
Load
CPU Core #5
OpenHardwareMonitor.Hardware.CPU.IntelCPU
Load
CPU Core #6
OpenHardwareMonitor.Hardware.CPU.IntelCPU
Load
CPU Total
OpenHardwareMonitor.Hardware.CPU.IntelCPU
Load
-------------------
ASRock Z390M-ITX/ac
Mainboard
-------------------
NVIDIA GeForce RTX 2070
GpuNvidia
GPU Core
OpenHardwareMonitor.Hardware.Nvidia.NvidiaGPU
Temperature
GPU Core
OpenHardwareMonitor.Hardware.Nvidia.NvidiaGPU
Clock
GPU Memory
OpenHardwareMonitor.Hardware.Nvidia.NvidiaGPU
Clock
GPU Shader
OpenHardwareMonitor.Hardware.Nvidia.NvidiaGPU
Clock
GPU Core
OpenHardwareMonitor.Hardware.Nvidia.NvidiaGPU
Load
GPU Memory Controller
OpenHardwareMonitor.Hardware.Nvidia.NvidiaGPU
Load
GPU Video Engine
OpenHardwareMonitor.Hardware.Nvidia.NvidiaGPU
Load
GPU Memory Total
OpenHardwareMonitor.Hardware.Nvidia.NvidiaGPU
SmallData
GPU Memory Used
OpenHardwareMonitor.Hardware.Nvidia.NvidiaGPU
SmallData
GPU Memory Free
OpenHardwareMonitor.Hardware.Nvidia.NvidiaGPU
SmallData
GPU Memory
OpenHardwareMonitor.Hardware.Nvidia.NvidiaGPU
Load
-------------------
ST1000DM010-2EP102
HDD
Temperature
OpenHardwareMonitor.Hardware.HDD.GenericHarddisk
Temperature
Used Space
OpenHardwareMonitor.Hardware.HDD.GenericHarddisk
Load
-------------------
Generic Memory
RAM
Memory
OpenHardwareMonitor.Hardware.RAM.GenericRAM
Load
Used Memory
OpenHardwareMonitor.Hardware.RAM.GenericRAM
Data
Available Memory
OpenHardwareMonitor.Hardware.RAM.GenericRAM
Data
```

DarthDestroyer, 2019-08-02:
Download the latest Version and use that lib ;-)

http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

Nova0ne, 2019-08-02:
> 
> 
> Download the latest Version and use that lib ;-)
> 
> http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

Hmmm... even when I use that library, I still am unable to fetch the temperature of the CPU. It isn't even showing as a sensor in the returned results.

DarthDestroyer, 2019-08-02:
Could you attach a Report from the latest OHM Version?

Nova0ne, 2019-08-02:
> 
> 
> Could you attach a Report from the latest OHM Version?

Here ya' go:

```
Open Hardware Monitor Report

--------------------------------------------------------------------------------

Version: 0.8.0.0

--------------------------------------------------------------------------------

Common Language Runtime: 4.0.30319.42000
Operating System: Microsoft Windows NT 6.2.9200.0
Process Type: 32-Bit

--------------------------------------------------------------------------------

Sensors

|
+- Intel Core i7-8700K (/intelcpu/0)
|  +- CPU Total      :       25       25       25 (/intelcpu/0/load/0)
|  +- CPU Core #1    :       50       50       50 (/intelcpu/0/load/1)
|  +- CPU Core #2    :        0        0        0 (/intelcpu/0/load/2)
|  +- CPU Core #3    :       50       50       50 (/intelcpu/0/load/3)
|  +- CPU Core #4    :       25       25       25 (/intelcpu/0/load/4)
|  +- CPU Core #5    :       25       25       25 (/intelcpu/0/load/5)
|  +- CPU Core #6    :        0        0        0 (/intelcpu/0/load/6)
|
+- ASRock Z390M-ITX/ac (/mainboard)
|
+- NVIDIA GeForce RTX 2070 (/nvidiagpu/0)
|  +- GPU Core       :      300      300      300 (/nvidiagpu/0/clock/0)
|  +- GPU Memory     :      405      405      405 (/nvidiagpu/0/clock/1)
|  +- GPU Shader     :        0        0        0 (/nvidiagpu/0/clock/2)
|  +- GPU Core       :       29       29       29 (/nvidiagpu/0/temperature/0)
|  +- GPU Core       :        6        6        6 (/nvidiagpu/0/load/0)
|  +- GPU Memory Controller :        5        5        5 (/nvidiagpu/0/load/1)
|  +- GPU Video Engine :        0        0        0 (/nvidiagpu/0/load/2)
|  +- GPU Memory     :   5.9433   5.9433   5.9433 (/nvidiagpu/0/load/3)
|  +- GPU Memory Free :  7705.13  7705.13  7705.13 (/nvidiagpu/0/smalldata/1)
|  +- GPU Memory Used :  486.875  486.875  486.875 (/nvidiagpu/0/smalldata/2)
|  +- GPU Memory Total :     8192     8192     8192 (/nvidiagpu/0/smalldata/3)
|
+- ST1000DM010-2EP102 (/hdd/0)
|  +- Temperature    :                            (/hdd/0/temperature/0)
|  +- Used Space     :                            (/hdd/0/load/0)
|
+- Generic Memory (/ram)
|  +- Memory         :                            (/ram/load/0)
|  +- Used Memory    :                            (/ram/data/0)
|  +- Available Memory :                            (/ram/data/1)

--------------------------------------------------------------------------------

Parameters

|
+- Intel Core i7-8700K (/intelcpu/0)
|
+- ASRock Z390M-ITX/ac (/mainboard)
|
+- NVIDIA GeForce RTX 2070 (/nvidiagpu/0)
|
+- ST1000DM010-2EP102 (/hdd/0)
|  |
[... 827 lines cut ...]
 07 Seek Error Rate                    4450A9000000 60    70    45    -
 09 Power-On Hours (POH)               6C0100000000 100   100   0     364
 0A Spin Retry Count                   000000000000 100   100   97    -
 0C Power Cycle Count                  5C0000000000 100   100   20    92
 B7 SATA Downshift Error Count         000000000000 100   100   0     0
 B8 End-to-End error                   000000000000 100   100   99    -
 BB Reported Uncorrectable Errors      000000000000 100   100   0     0
 BC Command Timeout                    000000000000 100   100   0     0
 BD High Fly Writes                    000000000000 100   100   0     -
 BE Temperature Difference from 100    1E00171E0000 65    70    40    30
 C1 Load Cycle Count                   5D0000000000 100   100   0     -
 C2 Temperature                        1E0000001600 22    30    0     30
 C3 Hardware ECC Recovered             EF664D020000 1     1     0     -
 C5 Current Pending Sector Count       000000000000 100   100   0     -
 C6 Uncorrectable Sector Count         000000000000 100   100   0     -
 C7 UltraDMA CRC Error Count           000000000000 200   200   0     -
 F0 Head Flying Hours                  6B010000EFCF 253   100   0     -
 F1 Total LBAs Written                 D7AF23630000 253   100   0     -
 F2 Total LBAs Read                    101588850000 253   100   0     -

Logical drive name: C:\
Format: NTFS
Total size: 999527804928
Total free space: 817870856192
```

DarthDestroyer, 2019-08-02:
Nope... That was the old Version 😉

Nova0ne, 2019-08-02:
> 
> 
> Nope... That was the old Version 😉

That's odd. I just returned that from the library you linked me.

DarthDestroyer, 2019-08-02:
that should be noted in the report, too! (it says point zero) 

maybe you have copied in the wrong direction?

Nova0ne, 2019-08-02:
> 
> 
> that should be noted in the report, too! (it says point zero)
> 
> maybe you have copied in the wrong direction?

Copied in the wrong direction... what do you mean? Incorrectly copied the output or my reference to my project?

DarthDestroyer, 2019-08-02:
you could only copy the .3 lib into your project...

my thought was, maybe you had copied the .0 Version into the .3 Folder! So, wrong direction 😉

Nova0ne, 2019-08-02:
> 
> 
> you could only copy the .3 lib into your project...
> 
> my thought was, maybe you had copied the .0 Version into the .3 Folder! So, wrong direction 😉

Ahh, now the output is clear. It must've been in the wrong direction, my bad. Let me see how this lib works and I'll get back to you.

Nova0ne, 2019-08-03:
All good, it seems. Issue resolved as far as I know. ;)

closed 2022-06-28 by D7EAD