openhardwaremonitor/openhardwaremonitor #1194: CPU Temperatures for Ryzen 7 3700X

FoxiesFury, 2019-07-31:
Missing temperatures for Ryzen 7 3700X. Here is report:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/3452986/OpenHardwareMonitor.Report.txt)

Thank you for your attention. :)

DarthDestroyer, 2019-08-02:
Did you tried #970?

FoxiesFury, 2019-08-04:
I tried it now. Does not work.

SearchForTheCode, 2019-08-06:
Can you try out the latest LibreHardwareMonitor?

FoxiesFury, 2019-08-06:
I downloaded and extracted it but how do I start it? :O

SearchForTheCode, 2019-08-06:
Oh you have no Visual Studio to build it? Please use this package to test. Just start the OpenHardwareMonitor.exe
[LibreOHM.zip](https://github.com/openhardwaremonitor/openhardwaremonitor/files/3472637/LibreOHM.zip)

FoxiesFury, 2019-08-06:
It works now! Thanks a lot!

SearchForTheCode, 2019-08-07:
Great news, so Ryzen 7 3700x is also handled by this AMD changes :)

FoxiesFury, 2019-08-07:
Screenshot and report:

[image]
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/3477810/OpenHardwareMonitor.Report.txt)

closed 2019-08-07 by FoxiesFury