openhardwaremonitor/openhardwaremonitor #1200: SENSOR_1 probe sensor not showing for ASUS Ranger VII matherboard.

Ndrinta, 2019-09-06:
Hello,
using the sw since ages and it has always worked beautifully. I now bought a probe sensor and added it on the SENSOR_1 probe port. The temperature value wasn't shown in the Open Hardware Monitor interface so I downloaded Fan Expert 3 and I could see that there is showed regularly so I assume there is no issue with the probe itself but rather a problem in the SW.

Is there any chance you can look into this? I would be glad to compensate you guys for the time spent on this fix! I'm serius.
Thanks

DarthDestroyer, 2019-09-06:
Could you attach a Report into your Issue?

That would help to identify the used Hardware!

Regards

Ndrinta, 2019-09-06:
Sure I can!

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/3585077/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2019-09-07:
You are a little bit outdated ;)

Have you tried the latest Version, too? 
http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

Ndrinta, 2019-09-08:
Yes I did yesterday. Since there was no more sensor shown I didn't bother to update.

http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

closed 2023-11-01 by Ndrinta