openhardwaremonitor/openhardwaremonitor #1203: AMD RX 5700 GPU Not showing any details

binary100100, 2019-09-19:
Version 0.8.0.3 Alpha shows the cards but no data.

Windows 10 Professional (ver 1809)
Adrenaline Driver 19.9.2

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/3629017/OpenHardwareMonitor.Report.txt)

sebastian-dev, 2019-09-23:
try out: https://github.com/aquacomputer/LibreHardwareMonitor/releases
or: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor

binary100100, 2019-09-23:
Nope. Same result. Nothing for RX 5700 XT cards.
[LibreHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/3644215/LibreHardwareMonitor.Report.txt)

moellerm, 2020-02-28:
I have started working on this. Please download this test version
http://openhardwaremonitor.org/files/openhardwaremonitor-v0.9.1.1-alpha.zip
and submit a report by selecting File -> Submit Report... from the menu.

moellerm, 2020-03-01:
This issue has been fixed in OHM version 0.9.2.

closed 2020-03-01 by moellerm