openhardwaremonitor/openhardwaremonitor #1208: Intel HD Graphics not listed

Ralbaker, 2019-10-06:
Hi,
I have a Lenovo G500 Laptop. It has a Intel HD graphics and a AMD graphics. The AMD graphics is not used only the Intel. 
Strangly only the AMD is listed with everything 0. The Intel graphics is not shown. Is this a bug or is there a setting I am missing?
Thanks
Best regards

DevilsJoker, 2019-12-18:
I have a similar issue on my Surface Book 2.
Only the NVIDIA card is listed, not the Intel HD graphics? (I do have correct values on the NVIDIA card)

Chris-LuYu, 2024-10-17:
I had the same situation, listing only the NVIDIA Graphics card information, but not the Intel(R) UHD Graphics 630 information. Does it not support the display of Intel graphics card information?