openhardwaremonitor/openhardwaremonitor #1211: HardwareMonitor wont start.

Fhosho, 2019-10-17:
This became a problem yesterday. (2019-10-15) 
I have HWM to start up with windows but get this error every time.. i try to start it as admin too but it's the same.. Tryed different versions also. 
[image]
Anyone else with the same problem or with a fix ? I miss my gadget!!

DarthDestroyer, 2019-10-17:
For the first step, i would rename the directory and start it as admin.

If that doesnt work, it could help to delete the config and start fresh... 

there is something wrong with the config and/or with the start procedure

Fhosho, 2019-10-17:
@DarthDestroyer 
hm, did both of your suggestions, did not seem to solve the problem, still the same.

DarthDestroyer, 2019-10-17:
so it doesnt even start normally?

Fhosho, 2019-10-17:
Nope.
Same error as the picture above.

Fhosho, 2019-10-18:
Okay so apparently nothing has worked for me.
But i found Open Hardware Monitor v0.8.0.5 ALPHA.
Wich at first didn't work either (at first)...
Got the same error as above but about one hour later when tried executing the same exe it magically opened up! so it's working now! 
(I hope it works when i reset my pc later) we will see about that.

closed 2019-10-18 by Fhosho

turbatus, 2019-12-30:
Same here, same error! 
[image]

Fhosho, 2019-12-30:
> Same here, same error!
> [image]

This error is now fixed for me, after some restarts of my pc It’s back to normal. Just wait and see if it launch normally after you restart your pc.

Tenn0, 2020-01-07:
For me, i configured OHW to startup with windows since i use it to monitor my temps in home assistant. But the above error appears also for me most of the time. But sometimes when my pc starts OHW starts correctly.

turbatus, 2020-01-26:
For me it works, now. Looks that it only needs some time to get to know your OS. :)

XcOM987, 2022-02-07:
[image]
Not sure if anyone has worked on this, but it's still an issue, I can replicate the fault on my server 2019 box with every version of Open Hardware available, reset the config, tried multiple config setups, compat mode, nothing works, get the same error every time