openhardwaremonitor/openhardwaremonitor #1213: CPU Temperature and clock speed and iGPU for i7-8850H

SebastiaanRutgers, 2019-10-24:
The CPU in my Laptop is only showing core load, no core speed or temps.
Also the iGPU is not showing up at all.
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/3767466/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2020-01-03:
Download the latest Version, please

http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

radfast, 2020-01-09:
Love this tool, thank you for writing and maintaining it.

Like the OP, I also have a Core i7-8850H.  What I'm seeing:

**Version: openhardwaremonitor-v0.8.0-beta**

- normally the CPU region shows the core load, no individual core speeds or temps
- exceptionally, one time when I used this version of the tool, it ran in a high detail mode with a lot more info.  I made no changes in settings, but I did make other recent changes to my system (new software AnVir Task Manager running for the first time, and I was making system changes to remove nvwmi64.exe which is a CPU + hard drive hog)

In the "high detail mode", it showed **all** the CPU details including individual clock speeds and temps for each core, and other temperature data like `distance to Tj max` for each core.  It also showed the nVidia GPU temperature (current and max). 

I didn't figure out how to reproduce this high detail mode again, though it was great!  Sorry, I did not take a screenshot.

----

**Version: openhardwaremonitor-v0.8.0-beta**

- CPU: temperatures for all cores, and package temperature
- CPU: no "distance to Tj max"
- nVidia GPU: no temperature shown

See screenshot.
[image]