openhardwaremonitor/openhardwaremonitor #1218: New Intel Pentium Silver CPUs

SearchForTheCode, 2019-11-20:
The new Intel Pentium Silver cpus in notebooks seem to be not working.

I uploaded an report for further analysis. In best case, this is just another detection case, no adjusting to any low level stuff.
[OpenHardwareMonitor.Report.1.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/3868849/OpenHardwareMonitor.Report.1.txt)

DarthDestroyer, 2019-12-03:
You know, that there is a newer Version or is it your fork?

A link for testing: http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

SearchForTheCode, 2019-12-03:
No thank you :) i did not see it. Currently i am using the LibreHardwareMonitor fork (as they do monthly updates). I will close this :) Thank you

closed 2019-12-03 by SearchForTheCode