openhardwaremonitor/openhardwaremonitor #1221: Empty Sensor Table In Web Server

adamgreschner, 2019-12-14:
All of my sensors populate in the application, however there are no sensors populated in the web server.  This is not a firewall issue because all firewalls are disabled and it is being accessed from the localhost machine.   

I have successfully configured the web sever on numerous machines

Please see the screenshot
[image]

Proplex, 2020-05-17:
I'm also experiencing this issue. Running version 0.9.2.0. Host is Windows 10 Pro.

Proplex, 2020-05-17:
I've also experienced this issue. After some digging, it turns out that a SSD of mine had `2.5"` in the official device name. That `"` broke JSON parsing in the web UI.

Check to see if you have any `"` or `'` in your device names. That might be breaking JSON parsing, which in turns breaks the web UI.