openhardwaremonitor/openhardwaremonitor #1225: Missing vcore, cpu fan and nvme temp

uglitangeo, 2020-01-03:
Missing the vcore (well, all of them, but this is the only one I need), CPU fan speed and nvme temp.

MSI Mortar Z370m
i7-9700K
XPG SX8200 PRO 1tb 

0.8.0.3 Alpha

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/4017879/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2020-01-03:
I think there is a Pull Request available for that Motherboard!

Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0xD352 at 0x4E/0x4F

uglitangeo, 2020-01-04:
Got the vcore and fan speed working, but nvme temps are still unavailable. I've only done basic coding for Arduino so this is above my skill set.

DarthDestroyer, 2020-01-04:
I would say that Arduino is heavier...
You just need Visual Studio and edit some Code at explained files and positions ;-)

Sorry, i missed the nvme part! 
There is a Pull Request available for nvme. 
 #984