openhardwaremonitor/openhardwaremonitor #1226: Nvidia GPU temp not displayed when GPU not in use

kingtongkt, 2020-01-08:
After eg. starting a game to use GPU, GPU temp still not displayed;
GPU temp is only displayed when GPU is currently in use when the program starts.

radfast, 2020-01-09:
I see this as well.  Maybe it's intended behaviour?  Probably the GPU cannot be polled for temperature when it is not currently in use.

I've noticed also if I stop using the GPU while openhardwaremonitor is running, the GPU temperature is then shown as 0.0 °C.

DarthDestroyer, 2020-01-09:
Dedicated GPUs are not available when not in use!

A workaround is available here: #952

ElleshaHackett, 2020-04-19:
I just connected Open Hardware Monitor to my home-assistant.io to show temperatures of my CPU, GPU and SSD on my dashboard. Unfortunately my GPU is indeed also only showing GPU temperature when used and otherwise 0.0 °C which is definitely incorrect. I would even prefer "unknown" instead of "0.0 °C" if it isn't possible. But rather I would of course like to see the real temperature. Can we fix this somehow? The workaround seems rather overkill.

I have a Dell XPS 9560 laptop with the GeForce GTX 1050 GPU.