openhardwaremonitor/openhardwaremonitor #1227: WMI requests stopped working. How to restore / reinstall or fix this error?

Artem-Mamzikov, 2020-01-28:
WMI requests stopped working. How to restore / reinstall or fix this error?
[image]

SkyEmie, 2020-06-26:
Sometimes I have the same problem, (and on librehardwaremonitor too), WMI requests don't work anymore

Artem-Mamzikov, 2020-08-19:
Внесу не большой ответ решение сам.
https://diyit.ru/viewtopic.php?p=150#p150

Возможно такой сбой может быть вызван некорректными командами в powershell причем все остальное может работать через wmi нормально.
При первом запуске создается компилируется и регистрируюца библиотеки C:\Windows\System32\wbem\Framework\root\OpenHardwareMonitor там лежат 2 файла Managed Object Format - MOF-файле возможно ошибка в них.
Чистка реестра полностью от OpenHardwareMonitor и nssm.exe а так же их поиск на ПК и дальней шее удаление потом установка с 0 не помогла, так же подгрузка недостающий кустов реестра связанных с OpenHardwareMonitor с рабочего ПК где все работает норм так же не помогла.

Дело было в wmi что то сбилось в нем на эту утилиту стал искать как заново создать параметры wmi

Восстановление работы Windows Management Instrumentation (WMI) после фатального сбоя

Устранение неполадок в WMI

Помогло
Проверяем целостность (На Windows XP и ниже не работает):

winmgmt /verifyrepository

В случае ответа отличного от “База данных WMI согласована”, можно выполнить "мягкое восстановление" командой:

winmgmt /salvagerepository

с последующим перезапуском службы:

net stop Winmgmt
net start Winmgmt

Если мягкое восстановление не помогло, пробуем вернуть хранилище в начальное состояние (последствия: все дополнительные классы WMI, когда-либо зарегистрированные в вашей системе, скорее всего, придётся регистрировать заново):

winmgmt /resetrepository

с последующим рестартом системы. Отмечу, что вторая машина заработала после этого этапа. Последствия были не сильно удручающими, но серьёзными: пришлось переинсталлировать Visual Studio и Delphi Starter, MS Office отказался работать и его пришлось деинсталлировать вручную, удаляя папки, файлы и записи из реестра, с последующей повторной установкой. Также слетели все наши собственные классы WMI.

Заново зарегистрировать модуль OpenHardwareMonitor после чего все заработало, но если у вас так же есть какие то сторонние модуля их так же нужно будет запустить для регистрации