openhardwaremonitor/openhardwaremonitor #1228: bug in openhardwaremonitor?

DonDarkore, 2020-01-29:
[image]

I just assembled new PC 

mobo: asrock b365 pro 4
cpu: intel i5 9400f
gpu: gtx 1660

As you can see on picture,  core nr 4 has 103 degrees and cpu fan 4 spins add 9820 rpm!   
I did not see him fly yet :)    He spins normally ….

On other monitor tool cpu core temps of core 4 is normal like others…..

Its my first time here on github, if i did something wrong by posting this, i'm sorry.
Just wanted to report bug.

moellerm, 2020-01-29:
The Temperature #4 is a temperature sensor from the mainboard, or something else connected to the super IO chip. It is not the temp of core 4.

Make sure not to run other monitoring tools in parallel with Open Hardware Monitor, especially official ones from the mainboard manufacturer, as the hardware access with them might not be synchronized. This might be the reason for the 9820 RPM fan speed.

You can add a report file here (File -> Save Report ... in OHM), but the chance that to find the reason for the above issues is very small. The main issue is lacking documentation of the mainboard circuits and the super IO chips.

DonDarkore, 2020-01-29:
> The Temperature #4 is a temperature sensor from the mainboard, or something else connected to the super IO chip. It is not the temp of core 4.
> Make sure not to run other monitoring tools in parallel [...]

But my cpu is not broken? right?  (i dont think so) 

  I will study your answer more deeper later. See what i can do and provide with better information.   I love opensource and free stuff. I have used openhardware for years…

closed 2020-01-30 by moellerm