openhardwaremonitor/openhardwaremonitor #1230: Option to delete logs after a set period of time.

MailYouLater, 2020-02-11:
I like to have OpenHardwareMonitor log my sensor data so I can refer to it later if there is a problem, but over time these logs start taking up a lot of space and I have to reclaim that space manually by going to the location where the logs are stored and deleting them.  I'd like to request an optional feature that allows me to set an amount of time that logs should be kept, after which OpenHardwareMonitor would automatically delete the old logs to keep from filling up my hard drive with them.

iNgeon, 2020-02-29:
Would love this and also be able to clear\reset the plot of all historical data to start fresh without doing a fresh install

nikolaysu, 2023-08-02:
I join the request
Or at least not block (close the descriptor) the old raw data so that they can be deleted by the script

iNgeon, 2023-08-02:
Appears this project was taken on by another dev and split into another branch, have a look at LibreHardwareMonitor which has this feature now

nikolaysu, 2023-08-03:
> Appears this project was taken on by another dev and split into another branch, have a look at LibreHardwareMonitor which has this feature now
Thanks, this works as it should.