openhardwaremonitor/openhardwaremonitor #1235: Please add support for Asus Maximus IV Extreme-Z

Banana-97, 2020-03-07:
There are sensors missing from OHM mainly opt-temp (temperature from optional temperature probe with is installed) and opt-fan RPM which are shown in HWinfo, and Ai Suite II but not OHM (all versions).

ASUS Maximus IV Extreme-Z is using Nuvoton NCT6776F chipset.

moellerm, 2020-03-07:
Can your run OHM and HWinfo in parallel, take a screenshot of HWinfo sensors page and save a report with OHM (File -> Save Report...)?
I can take a look if there is something obvious missing in OHM.

Banana-97, 2020-03-08:
> Can your run OHM and HWinfo in parallel, take a screenshot of HWinfo sensors page and save a report with OHM (File -> Save Report...)?
> I can take a look if there is something obvious missing in OHM.

Here are the files, I have included HWiNFO report too if that helps. Sensors which are missing is under the ASUS Maximus IV Extreme-Z (Asus ROG) tab on HWiNFO.

[HWiNFO Log.LOG](https://github.com/openhardwaremonitor/openhardwaremonitor/files/4302950/HWiNFO.Log.LOG)
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/4302951/OpenHardwareMonitor.Report.txt)
[image]

moellerm, 2020-03-08:
Thank you for the report and screenshot. Unfortunately I have no idea where to get the "ASUS ROG" sensor values. In order to implement this we need some documentation about the hardware (chip type) and/or the data access interface used. 

I'm closing this issue, because it can't be fixed without the missing info. If any clue about the "ASUS ROG" part shows up, don't hesitate to post here, and I will reopen the issue.

closed 2020-03-08 by moellerm