openhardwaremonitor/openhardwaremonitor #1236: Incorrect Radeon Vega 8 GPU usage

cyberixae, 2020-03-08:
Open Hardware Monitor reports the GPU load of my Radeon Vega 8 as 100% at all times. Windows task manager seems to report more appropriate values. See attached screenshot.
[image]

moellerm, 2020-03-08:
Can you submit a report (File -> Submit Report ...)?

cyberixae, 2020-03-08:
Done.

moellerm, 2020-03-08:
Thank you. The results on your system are rather strange. The values for AMD GPUs are read from the ADL (AMD Display Library). That's the official AMD interface to get values from the GPU driver. The library claims that the Overdrive feature set is not supported and not enabled. At the same time the function ADL_Overdrive5_CurrentActivity_Get used to get the core/memory clock, core voltage and core load returns OK as status. But the values (beside perhaps the clocks) seem not very correct.

Can you try and see what GPU-Z can read on your system? Please post a screenshot of the "Sensors" tab.

cyberixae, 2020-03-08:
Here you go

[image]

moellerm, 2020-03-08:
Thank you. I have compiled a new alpha version with more details in the report for AMD GPUs. Can you run this new version

https://openhardwaremonitor.org/files/openhardwaremonitor-v0.9.2.1-alpha.zip

and submit once more a report? This version should still display the same values in the interface, but give more detail in the report. I want to make sure with the new report that I haven't missed anything obvious.

cyberixae, 2020-03-08:
Done.

moellerm, 2020-03-09:
Thank you. Unfortunately, the ADL functions currently used in Open Hardware Monitor do not provide any valid GPU load data. I have extended once more the report details for ADL functions on AMD GPUs.

https://openhardwaremonitor.org/files/openhardwaremonitor-v0.9.2.2-alpha.zip

Could you submit once more a report with this new version? I hope to at least mask the invalid GPU load on hardware like yours (but this is not implemented yet in v0.9.2.2-alpha, need more input first).

cyberixae, 2020-03-10:
Done.

moellerm, 2020-03-11:
Thx. To me this looks like a bug in the ADL or the AMD GPU driver. I have added additional checks that should prevent impossible GPU loads from getting displayed. Unfortunately this does not guarantee that all erroneous values are caught. Wrong values in the interval 0-100 are still displayed. But atm I do not have any other method. The ADL claims the usage value to be supported and correct while it is obviously wrong. I do not know on which hardware and/or driver version this problem occurs, so I can't filter based on those. Here is a new version which should hide the load sensor on your system (and additionally print the driver version in the report).

https://openhardwaremonitor.org/files/openhardwaremonitor-v0.9.2.3-alpha.zip 

I do not know what method GPU-Z uses for getting the GPU load value.

closed 2020-03-29 by moellerm