openhardwaremonitor/openhardwaremonitor #1237: Gadget Bars Don't Scale

dallas7, 2020-03-10:
i9-9900K Z390 Win10pro OHM 0.9.2

The clocks CPU core bars in the gadget aren't representative of the values.

In the screenshot, they're always those tiny levels on the left no matter what the MHz is.  The shot was taken during the conversion of a 32GB m2ts video to 4GB.  Surely the bars should span about half way.

Thanks!

[image]

moellerm, 2020-03-10:
The clocks should be displayed as "2600 MHz" and not as bars. The bars are for the load sensors.

closed 2020-03-10 by moellerm

dallas7, 2020-03-10:
Thanks for the fast reply.

It's been a long time since I've dug into the Gadget options and got my CPU Core# "show in" mixed up.